mavlink/MAVSDK #1282: mavsdk_server does not clean mission state after flight

petergerten, 2020-12-05:
If I want to start a second mission after a successful first mission - without restarting mavsdk_server -
the previous state persists.

Resulting in things like: is_mission_finished being true.
This happens even after clear_mission() and reuploading a new mission.

Is there any "cleanup" action I have to do that I am not aware about?

julianoes, 2020-12-05:
Thanks for the issue. That doesn't sound like the correct behaviour. Which PX4 version are you using? And can you reproduce this against SITL?

petergerten, 2020-12-05:
I am currently experiencing this with a customized PX4 firmware on the Yuneec H520 (in HITL). Will try in SITL with standard firmware now.

petergerten, 2020-12-05:
I can reproduce it in SITL using PX4 Release 1.10.1

julianoes, 2020-12-05:
@petergerten does it still happen with 1.11?

L4ncelot, 2021-02-02:
@julianoes, I can confirm the same behavior with 1.11 in the SITL configuration.

julianoes, 2021-02-03:
@L4ncelot would be great if you could test #1304.

L4ncelot, 2021-02-03:
@julianoes #1304 works great, thank you for the support.

PR #1304 (2021-02-03): Reset mission progress on clear and mission upload
Fixes #1282
closed 2021-02-03 by JonasVautherin