mavlink/MAVSDK #1292: "Cannot open source file plugin_base.h" error

R-Paladugu, 2021-01-10:
As the title suggests, I keep getting a weird error as stated above that prevents me from using any of the mavsdk api because I cannot include it into my programs. I checked to make sure everything was installed correctly and nothing seemed broken. How can I fix this?

I am on an Ubuntu 20.04 VM if that helps.

R-Paladugu, 2021-01-10:
Update: I can use the source files that aren't included in the plugins folder, everything located before that folder doesn't give me the same issue.

JonasVautherin, 2021-01-10:
I cannot really guess what you are trying to do. Did you try to to build the examples? Did you succeed?

R-Paladugu, 2021-01-10:
I ended up figuring out the issue, it was related to my IDE not picking up the MAVSDK folder rather than MAVSDK not being installed properly.

closed 2021-01-11 by JonasVautherin