mavlink/MAVSDK #1296: Compilation failed for Raspberry pi with undefined reference to `__atomic_store_8'

Katawann, 2021-01-29:
When I try to compile MAVSDK C++ on my Raspberry Pi (RPi 3B+ armv7l with Raspbian GNU/Linux 10 buster), it fails with the `v0.35.1` but not with the `v0.33.1`

With the `v0.35.1`, it finishes with the error below:

```
$ cmake -DCMAKE_BUILD_TYPE=Debug -DBUILD_SHARED_LIBS=ON -DENABLE_MAVLINK_PASSTHROUGH=ON -DBUILD_TEST=ON -Bbuild -H.

...

-- Build files have been written to: /home/pi/libs/MAVSDK/build
$ sudo cmake --build build

...

[ 39%] Building CXX object src/CMakeFiles/unit_tests_runner.dir/plugins/telemetry/math_conversions_test.cpp.o
[ 40%] Linking CXX executable unit_tests_runner
/usr/bin/ld: core/libmavsdk.so.0.35.1: undefined reference to `__atomic_store_8'
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/unit_tests_runner.dir/build.make:349: src/unit_tests_runner] Error 1
make[1]: *** [CMakeFiles/Makefile2:127: src/CMakeFiles/unit_tests_runner.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

julianoes, 2021-02-01:
You need something like https://github.com/mavlink/MAVSDK/blob/e869ba85909d39d69ad4b326e41c9378badd1417/src/core/CMakeLists.txt#L74 but it should only be added for armv6 and armv7 targets, so it would need some if to check for that. Would you mind trying that out and making a pull request once you find a solution that works?

Katawann, 2021-02-01:
Done ! I don't have the issue anymore with the check thank you

closed 2021-02-01 by Katawann

JonasVautherin, 2021-02-01:
Thank you for the contribution!