mavlink/MAVSDK #1305: errors while build MAVSDK on raspberry pi 3

helloE9, 2021-02-03:
Hi,
I got this error while I am building last stable branch of MAVSDK.

pi@raspberrypi:~/MAVSDK $ cmake --build build/default
[  1%] Built target gtest
[ 18%] Built target mavsdk
[ 20%] Built target mavsdk_calibration
[ 23%] Built target mavsdk_camera
[ 24%] Built target mavsdk_mission
[ 26%] Built target mavsdk_telemetry
[ 27%] Built target gmock
[ 28%] Built target gtest_main
[ 29%] Linking CXX executable unit_tests_runner
/usr/bin/ld: core/libmavsdk.so.0.35.1: undefined reference to `__atomic_store_8'
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/unit_tests_runner.dir/build.make:349: src/unit_tests_runner] Error 1
make[1]: *** [CMakeFiles/Makefile2:127: src/CMakeFiles/unit_tests_runner.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

julianoes, 2021-02-03:
This should be resolved in the latest develop, see https://github.com/mavlink/MAVSDK/pull/1300.

closed 2021-02-03 by julianoes

helloE9, 2021-02-04:
At this time, I got this error,
after I did, 
$ cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=ON -Bbuild/default -H.
...
Install the project...
-- Install configuration: "Release"
-- Installing: /home/pi/MAVSDK/build/default/third_party/install/bin/curl-config
-- Up-to-date: /home/pi/MAVSDK/build/default/third_party/install/lib/pkgconfig/libcurl.pc
-- Up-to-date: /home/pi/MAVSDK/build/default/third_party/install/include/curl
-- Up-to-date: /home/pi/MAVSDK/build/default/third_party/install/include/curl/urlapi.h
-- Up-to-date: /home/pi/MAVSDK/build/default/third_party/install/include/curl/mprintf.h
-- Up-to-date: /home/pi/MAVSDK/build/default/third_party/install/include/curl/curlver.h
-- Up-to-date: /home/pi/MAVSDK/build/default/third_party/install/include/curl/curl.h
-- Up-to-date: /home/pi/MAVSDK/build/default/third_party/install/include/curl/multi.h
-- Up-to-date: /home/pi/MAVSDK/build/default/third_party/install/include/curl/easy.h
-- Up-to-date: /home/pi/MAVSDK/build/default/third_party/install/include/curl/options.h
-- Up-to-date: /home/pi/MAVSDK/build/default/third_party/install/include/curl/system.h
-- Up-to-date: /home/pi/MAVSDK/build/default/third_party/install/include/curl/stdcheaders.h
-- Up-to-date: /home/pi/MAVSDK/build/default/third_party/install/include/curl/typecheck-gcc.h
-- Up-to-date: /home/pi/MAVSDK/build/default/third_party/install/lib/cmake/CURL/CURLTargets.cmake
-- Installing: /home/pi/MAVSDK/build/default/third_party/install/lib/cmake/CURL/CURLTargets-release.cmake
-- Up-to-date: /home/pi/MAVSDK/build/default/third_party/install/lib/cmake/CURL/CURLConfigVersion.cmake
-- Up-to-date: /home/pi/MAVSDK/build/default/third_party/install/lib/cmake/CURL/CURLConfig.cmake
-- Installing: /home/pi/MAVSDK/build/default/third_party/install/lib/libcurl.a
-- Installing: /home/pi/MAVSDK/build/default/third_party/install/bin/curl
[ 62%] Completed 'curl'
[100%] Built target curl
-- Found curl via config file: /home/pi/MAVSDK/build/default/third_party/install/lib/cmake/CURL
CMake Error at src/CMakeLists.txt:65 (add_subdirectory):
  The source directory

    /home/pi/MAVSDK/src/third_party/gtest

  does not contain a CMakeLists.txt file.

-- BUILD_MAVSDK_SERVER not set: not building grpc mavsdk_server
-- Configuring incomplete, errors occurred!
See also "/home/pi/MAVSDK/build/default/CMakeFiles/CMakeOutput.log".
See also "/home/pi/MAVSDK/build/default/CMakeFiles/CMakeError.log".
---------------
which is the correct version for Raspbian?
I tried to build latest develop ver. of MAVSDK.

helloE9, 2021-02-04:
Hi, 
Why I still have this problem on Rasbian OS on Raspbery Pi 3?

pi@raspberrypi:/MAVSDK $ sudo cmake --build build/default
Scanning dependencies of target gtest
[  1%] Building CXX object src/third_party/gtest/googletest/CMakeFiles/gtest.dir/src/gtest-all.cc.o
[  1%] Linking CXX shared library ../../../../lib/libgtestd.so
[  1%] Built target gtest
Scanning dependencies of target mavsdk
[  2%] Building CXX object src/core/CMakeFiles/mavsdk.dir/call_every_handler.cpp.o
[  2%] Building CXX object src/core/CMakeFiles/mavsdk.dir/connection.cpp.o
In file included from /MAVSDK/src/core/mavsdk_impl.h:13:0,
                 from /MAVSDK/src/core/connection.cpp:2:
/MAVSDK/src/core/safe_queue.h:5:20: fatal error: optional: No such file or directory
compilation terminated.
make[2]: *** [src/core/CMakeFiles/mavsdk.dir/build.make:76: src/core/CMakeFiles/mavsdk.dir/connection.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:182: src/core/CMakeFiles/mavsdk.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

// 
is there I should clone the correct release for Raspberry Pi?