mavlink/MAVSDK #1309: Build a Raspberry companion for PX4 to use MAVSDK

helloE9, 2021-02-04:
Hi,
Why I still have this problem on Rasbian OS on Raspbery Pi 3?

pi@raspberrypi:/MAVSDK $ sudo cmake --build build/default
Scanning dependencies of target gtest
[ 1%] Building CXX object src/third_party/gtest/googletest/CMakeFiles/gtest.dir/src/gtest-all.cc.o
[ 1%] Linking CXX shared library ../../../../lib/libgtestd.so
[ 1%] Built target gtest
Scanning dependencies of target mavsdk
[ 2%] Building CXX object src/core/CMakeFiles/mavsdk.dir/call_every_handler.cpp.o
[ 2%] Building CXX object src/core/CMakeFiles/mavsdk.dir/connection.cpp.o
In file included from /MAVSDK/src/core/mavsdk_impl.h:13:0,
from /MAVSDK/src/core/connection.cpp:2:
/MAVSDK/src/core/safe_queue.h:5:20: fatal error: optional: No such file or directory
compilation terminated.
make[2]: *** [src/core/CMakeFiles/mavsdk.dir/build.make:76: src/core/CMakeFiles/mavsdk.dir/connection.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:182: src/core/CMakeFiles/mavsdk.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

//
is there I should clone the correct release for Raspberry Pi?

JonasVautherin, 2021-02-04:
First, `cmake` does not need `sudo`. So this is most likely wrong: `sudo cmake --build build/default`.

Can you show the command line you use for the configure step? Is that a clean build?

julianoes, 2021-02-04:
>  fatal error: optional: No such file or directory

It looks like your compiler and standard library are too old and don't support features (like `std::optional`) used in MAVSDK.
What RPi OS image are you using? You probably have to update to something newer.

helloE9, 2021-02-04:
> First, `cmake` does not need `sudo`. So this is most likely wrong: `sudo cmake --build build/default`.
yes. after got some error several times, I just thought to try with 'sudo' if there was any permission problem.

> Can you show the command line you use for the configure step? Is that a clean build?
I did follow steps listed on build guide from MADSDK github 

sudo apt-get update -y
sudo apt-get install cmake build-essential colordiff git doxygen -y

Clone the MAVSDK repository (or your fork):
git clone https://github.com/mavlink/MAVSDK.git
cd MAVSDK
git checkout develop
git submodule update --init --recursive

cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=ON -Bbuild/default -H.
cmake --build build/default
->>> at this phase I've got some errors

and, as julianoes has mentioned bellow, I forgot I installed old gcc to test very old module :(
so, now I did upgrade gcc to 8.3 and doing build again.

helloE9, 2021-02-04:
> > fatal error: optional: No such file or directory
> 
> It looks like your compiler and standard library are too old and don't support features (like `std::optional`) used in MAVSDK.
> What RPi OS image are you using? You probably have to update to something newer.

Thanks!!! and, I forgot I installed old gcc to test very old module :(
so, now I did upgrade gcc to 8.3 and doing build again.

julianoes, 2021-02-04:
Nice, so the issue can be closed?

helloE9, 2021-02-05:
e> Nice, so the issue can be closed?

Thanks, 
Everything goes well :) 
Wish happy weekend for you guys.

closed 2021-02-05 by helloE9