mavlink/MAVSDK #1313: MAVSDK server build fails

DoppleGangster, 2021-02-08:
I'm trying to build on an NVIDIA Xavier NX. I have managed to build the server previously, however I wanted to use the updated build. It looks like it wants me to rebuild some of the libraries using -fPIC? Not sure if this is the right direction to start looking or if I have a bigger problem.

I followed the instructions [here](https://mavsdk.mavlink.io/develop/en/contributing/build.html#install-artifacts) building from source and I get the following error:

[ 94%] Linking CXX shared library libmavsdk_server.so
/usr/bin/ld: ../../../third_party/install/lib/libcrypto.a(sha1-armv8.o): relocation R_AARCH64_PREL64 against symbol `OPENSSL_armcap_P' which may bind externally can not be used when making a shared object; recompile with -fPIC
../../../third_party/install/lib/libcrypto.a(sha1-armv8.o): In function `sha1_block_armv8':
(.text+0x1240): dangerous relocation: unsupported relocation
/usr/bin/ld: ../../../third_party/install/lib/libcrypto.a(chacha-armv8.o): relocation R_AARCH64_PREL64 against symbol `OPENSSL_armcap_P' which may bind externally can not be used when making a shared object; recompile with -fPIC
../../../third_party/install/lib/libcrypto.a(chacha-armv8.o):(.text+0x20): dangerous relocation: unsupported relocation
/usr/bin/ld: ../../../third_party/install/lib/libcrypto.a(poly1305-armv8.o): relocation R_AARCH64_ADR_PREL_LO21 against symbol `poly1305_blocks' which may bind externally can not be used when making a shared object; recompile with -fPIC
../../../third_party/install/lib/libcrypto.a(poly1305-armv8.o): In function `poly1305_init':
(.text+0x40): dangerous relocation: unsupported relocation
/usr/bin/ld: ../../../third_party/install/lib/libcrypto.a(poly1305-armv8.o): relocation R_AARCH64_ADR_PREL_LO21 against symbol `poly1305_emit' which may bind externally can not be used when making a shared object; recompile with -fPIC
(.text+0x48): dangerous relocation: unsupported relocation
/usr/bin/ld: ../../../third_party/install/lib/libcrypto.a(poly1305-armv8.o): relocation R_AARCH64_PREL64 against symbol `OPENSSL_armcap_P' which may bind externally can not be used when making a shared object; recompile with -fPIC
../../../third_party/install/lib/libcrypto.a(poly1305-armv8.o): In function `poly1305_emit_neon':
(.text+0x9a0): dangerous relocation: unsupported relocation
/usr/bin/ld: ../../../third_party/install/lib/libcrypto.a(sha256-armv8.o): relocation R_AARCH64_PREL64 against symbol `OPENSSL_armcap_P' which may bind externally can not be used when making a shared object; recompile with -fPIC
../../../third_party/install/lib/libcrypto.a(sha256-armv8.o): In function `sha256_block_data_order':
(.text+0xf88): dangerous relocation: unsupported relocation
/usr/bin/ld: ../../../third_party/install/lib/libcrypto.a(sha512-armv8.o): relocation R_AARCH64_PREL64 against symbol `OPENSSL_armcap_P' which may bind externally can not be used when making a shared object; recompile with -fPIC
../../../third_party/install/lib/libcrypto.a(sha512-armv8.o): In function `sha512_block_data_order':
(.text+0x1108): dangerous relocation: unsupported relocation
collect2: error: ld returned 1 exit status
src/mavsdk_server/src/CMakeFiles/mavsdk_server.dir/build.make:226: recipe for target 'src/mavsdk_server/src/libmavsdk_server.so' failed
make[2]: *** [src/mavsdk_server/src/libmavsdk_server.so] Error 1
CMakeFiles/Makefile2:1992: recipe for target 'src/mavsdk_server/src/CMakeFiles/mavsdk_server.dir/all' failed
make[1]: *** [src/mavsdk_server/src/CMakeFiles/mavsdk_server.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

Thanks!

JonasVautherin, 2021-02-08:
Can you try adding something like `CFLAGS="-fPIC"` to [this line](https://github.com/mavlink/MAVSDK/blob/develop/third_party/openssl/CMakeLists.txt#L69)? Something like:

```
CONFIGURE_COMMAND ${CMAKE_COMMAND} -E env CROSS_COMPILE="" CFLAGS="-fPIC" <SOURCE_DIR>/Configure ${OPENSSL_PLATFORM} --prefix=${CMAKE_INSTALL_PREFIX} no-shared
```

DoppleGangster, 2021-02-09:
Still getting the same error with the change.

DoppleGangster, 2021-02-09:
Sorry should have added that I am building local on the machine. I did update the case that would correspond with this, at least I believe.

I'll try cross compiling in the morning to see if I get a more desired result. 

Thanks again for your input.

JonasVautherin, 2021-02-09:
Ah... would it be [here](https://github.com/mavlink/MAVSDK/blob/develop/third_party/openssl/CMakeLists.txt#L86), then?

DoppleGangster, 2021-02-09:
It was there. The cross compile worked so think I am good.

Thanks!

closed 2021-02-09 by DoppleGangster