mavlink/MAVSDK #1316: Getting "timeout" issue while uploading mission on PixHawk (PX4) via MAVSDK

bb17pugc, 2021-02-10:
Hi,
I am using MAVSDK to upload mission from terminal to my PixHawk 4 (PX4) via telemetry.
Sometimes it works like a charm and sometime it give the "timout" error and stops.

Can someone please tell the fix for this issue? I tired to go into the file and updated the timeout value from "none" to "50" but it didn't worked. Same issue, sometimes it work and sometimes not work.

Complete code file is attached here. (due to editor restriction, changed the extension from ".py" to ".txt").
I am using Ubuntu 20.04

[image]

[mission.txt](https://github.com/mavlink/MAVSDK/files/5957145/mission.txt)

julianoes, 2021-02-10:
Your code looks all fine to me.

I'm suspecting a problem with the connection. Are you using a radio or a direct FTDI cable?
And which PX4 version are you using?

Maybe we could resolve this by just increasing the timeout or making it configurable.

bb17pugc, 2021-02-10:
I am using the **FTDI Cable** to connect the Telemetry with my laptop.
Telemetry: **RFD900+**

PX4 Version: **1.11**
Where we can increase that time and what is the unit (sec/msec) of time to increase?

julianoes, 2021-02-10:
Ok, and is the upload working when the drone is close, and presumably the connection is good?

julianoes, 2021-02-10:
The timeout is set in https://github.com/mavlink/MAVSDK/blob/855a4b5d1e7c13a601781792f25dc5fe08ff5ae3/src/core/mavlink_mission_transfer.h#L248 so you would have to rebuild mavsdk_server yourself.

bb17pugc, 2021-02-11:
> Ok, and is the upload working when the drone is close, and presumably the connection is good?

Drone is at about 3-4 feet from the GCS telemetry. Even then the problem is coming.
Sometime it works with the 200+ mission items, and sometimes it didn't work even with the 5 mission items.

I will try the change in timeout and then let you know the feedback. Thank you.
What is the unit of time we are giving? Is this seconds or something else?

julianoes, 2021-02-11:
It's seconds there. Are you rebuilding mavsdk_server yourself already?

bb17pugc, 2021-02-11:
> It's seconds there. Are you rebuilding mavsdk_server yourself already?

Yes, I am on it as you directed, but not doing anything before. But getting problems while building that.
I am using [this](https://mavsdk.mavlink.io/develop/en/contributing/build.html) guide to build that with our timeout changes.

julianoes, 2021-02-12:
What architecture are you on? I can build a binary and upload that for you.

MuhammadBilal1, 2021-02-12:
> What architecture are you on? I can build a binary and upload that for you.

That will be great if you can please do this favour.

I am using Ubuntu 20.04 for doing all this.

julianoes, 2021-02-12:
That's x86-64? That should actually be fairly straightforward to build then, just do:

```
cmake -Bbuild/default -DBUILD_MAVSDK_SERVER=ON -DSUPERBUILD=ON .
cmake --build build/default -j8
```

MuhammadBilal1, 2021-02-12:
I am using x64.

julianoes, 2021-02-12:
[mavsdk_server.zip](https://github.com/mavlink/MAVSDK/files/5971918/mavsdk_server.zip)

try this

MuhammadBilal1, 2021-02-12:
Where I put this file?

> [mavsdk_server.zip](https://github.com/mavlink/MAVSDK/files/5971918/mavsdk_server.zip)
> 
> try this

Where I put this file?

julianoes, 2021-02-12:
You have to run it with:
```
./mavsdk_server -p 50051 serial:///dev/ttyUSB0:57600
```

And then in your script you have to use:
```
drone = System(mavsdk_server_address="localhost", port=50051)
await drone.connect()
```

The script and the binary have to run on the same machine.

MuhammadBilal1, 2021-02-16:
> so give you a bit more debug output

I tried to follow the instructions you gave, and attached are the results. So far no success in running the operation.

[image]
[image]
[image]

julianoes, 2021-02-16:
Hm, I can see how it looks busy because it's trying to get several parameters, etc. all at startup. Does it work better if you wait a couple of seconds until these things have stabilized before trying to upload a mission?

julianoes, 2021-03-17:
@MuhammadBilal1 @bb17pugc were you able to resolve this?

I'm planning to add a parameter for overall mavsdk to set the timeout.

MuhammadBilal1, 2021-03-17:
> @MuhammadBilal1 @bb17pugc were you able to resolve this?
> 
> I'm planning to add a parameter for overall mavsdk to set the timeout.

No, I wasn't able to resolve this issue. Adding the parameter will be helpful. Thank you.
I will be waiting :-)

julianoes, 2021-03-18:
@MuhammadBilal1 see #1362

philippecloutiermathura, 2021-08-12:
Hi, I'm getting the same issue while connecting with a USB antenna (serial connection).  When uploading a mission with more than two mission items, it times out almost immediately.  I tried setting the timeout value to "2" or "5" but I get the same result.  I also tried using the MAVSDK-Python project example [mission.py](https://github.com/mavlink/MAVSDK-Python/blob/main/examples/mission.py) and it times out  (however, if I send only one or two mission items, it works).  
I also tried connecting to the drone using a wifi hotspot and uploading a mission (using tcp).  Using that setup, the timeout issue does not occur.

MAVSDK-Python: 0.19.0
PX4: 1.11.3
Python: 3.8.10
Drone connection: USB Antenna 

```
    await drone.connect(system_address="serial:///dev/ttyUSB0:57600")
    await drone.core.set_mavlink_timeout(2)
```

```
  File "/home/xgl-gs1/xguardlabs-uav-controller/mission_test.py", line 129, in run
    await drone.mission.upload_mission(mission_plan)
  File "/home/xgl-gs1/xguardlabs-uav-controller/venv/lib/python3.8/site-packages/mavsdk/mission.py", line 707, in upload_mission
    raise MissionError(result, "upload_mission()", mission_plan)
mavsdk.mission.MissionError: TIMEOUT: 'Timeout'; origin: upload_mission(); params: (<mavsdk.mission.MissionPlan object at 0x7f2d36405460>,)
```

JonasVautherin, 2021-08-14:
There have been some improvements in the latest version of mavsdk (coming with MAVSDK-C++ 0.42.0), and also in PX4. I think it will be worth testing again with those versions when they are out.

philippecloutiermathura, 2021-08-20:
Thank you for the suggestion, I will try again when a more recent version of MAVSDK-Python is available.  Correct me if I am wrong, but I think the current MAVSDK-Python 0.19.0 (released july 23) does not contain the changes from MAVSDK-C++ 0.42.0 (released august 6) .  I will also try updating my PX4 version.

julianoes, 2021-08-30:
If you have troubles uploading missions, you might also want to try this: https://github.com/PX4/PX4-Autopilot/issues/18082

julianoes, 2024-05-27:
Closing this as there was no more follow up but plenty of fixes in the meantime.

closed 2024-05-27 by julianoes

NabinBhandari1, 2025-04-24:
Hello @julianoes  I am facing the same problem while uploading and downloading the mission file. Currently I'm using raspberry pi5. for running the script. my script work fine while running in laptop using UDP connection. But when I run the came script in pi5 changing the connection type to serial dev/ttyAMA0 i receive time out error. I also tried to build the custom mavsdk server as mentioned by you by changing  "static constexpr double timeout_s = 2"  still I reseive the time out error. I think I did some mistake while building the server. Could help me build the server for Raspberrry pi =5 architecture ?

julianoes, 2025-04-27:
@NabinBhandari1 please don't necrobump. Create a new issue and give all the context, versions, things you tried, logs, etc..