mavlink/MAVSDK #1318: Include MAVSDK in QtCreator CMAKE project

YogeshHC, 2021-02-11:
I'm trying to build a custom application with MAVSDK. I've built MAVSDK following the procedure mentioned [here ](https://mavsdk.mavlink.io/develop/en/contributing/build.html) and I'm able to compile and build [examples ](https://github.com/mavlink/MAVSDK/tree/develop/examples/battery) using CMAKE. However, when I try to include MAVSDK in my application(QT creator with CMAKE) i get ``` undefined reference to `mavsdk::Mavsdk::Mavsdk()' ```

```
cmake_minimum_required(VERSION 3.5)

project(WGCS LANGUAGES CXX)

set(CMAKE_INCLUDE_CURRENT_DIR ON)

set(CMAKE_AUTOUIC ON)
set(CMAKE_AUTOMOC ON)
set(CMAKE_AUTORCC ON)

set(CMAKE_CXX_STANDARD 11)
set(CMAKE_CXX_STANDARD_REQUIRED ON)

# QtCreator supports the following variables for Android, which are identical to qmake Android variables.
# Check http://doc.qt.io/qt-5/deployment-android.html for more information.
# They need to be set before the find_package(Qt5 ...) call.

#if(ANDROID)
#    set(ANDROID_PACKAGE_SOURCE_DIR "${CMAKE_CURRENT_SOURCE_DIR}/android")
#    if (ANDROID_ABI STREQUAL "armeabi-v7a")
#        set(ANDROID_EXTRA_LIBS
#            ${CMAKE_CURRENT_SOURCE_DIR}/path/to/libcrypto.so
#            ${CMAKE_CURRENT_SOURCE_DIR}/path/to/libssl.so)
#    endif()
#endif()

find_package(QT NAMES Qt6 Qt5 COMPONENTS Widgets REQUIRED)
find_package(Qt${QT_VERSION_MAJOR} COMPONENTS Widgets REQUIRED)

if(NOT MSVC)
    add_definitions("-Wall -Wextra ")
    # This warning is triggered by the MAVLink includes.
    add_definitions("-Wno-address-of-packed-member")
else()
    add_definitions("-WX -W2")
endif()

find_package(MAVSDK REQUIRED)

include_directories("C:/Program Files/MAVSDK/include")
link_directories("C:/Program Files/MAVSDK/lib")

if(ANDROID)
  add_library(WGCS SHARED
    main.cpp
    mainwindow.cpp
    mainwindow.h
    mainwindow.ui
  )
else()
  add_executable(WGCS
    main.cpp
    mainwindow.cpp
    mainwindow.h
    mainwindow.ui
  )
endif()

target_link_libraries(WGCS PUBLIC
    mavsdk
    Qt${QT_VERSION_MAJOR}::Widgets
    )
```
main.cpp in Qt Creator
```
#include "mainwindow.h"
#include <QApplication>
#include <mavsdk/mavsdk.h>
#include <chrono>
#include <cstdint>
#include <iostream>
#include <future>
#include <memory>

using namespace mavsdk;

int main(int argc, char *argv[])
{
    QApplication a(argc, argv);
    MainWindow w;
    Mavsdk mavsdk;
    w.show();
    return a.exec();
}
```
Error
```
19:14:17: Running steps for project WGCS...
19:14:17: Starting: "C:\Qt\Tools\CMake_64\bin\cmake.exe" --build . --target all
[1/3 13.4/sec] Automatic MOC and UIC for target WGCS
[2/3 2.2/sec] Building CXX object CMakeFiles/WGCS.dir/WGCS_autogen/mocs_compilation.cpp.obj
[3/3 2.9/sec] Linking CXX executable WGCS.exe
FAILED: WGCS.exe 
cmd.exe /C "cd . && C:\Qt\Tools\mingw810_64\bin\g++.exe -O3 -DNDEBUG  CMakeFiles/WGCS.dir/WGCS_autogen/mocs_compilation.cpp.obj CMakeFiles/WGCS.dir/main.cpp.obj CMakeFiles/WGCS.dir/mainwindow.cpp.obj -o WGCS.exe -Wl,--out-implib,libWGCS.dll.a -Wl,--major-image-version,0,--minor-image-version,0 -LC:/PROGRA~1/MAVSDK/lib -lmavsdk  C:/Qt/5.15.2/mingw81_64/lib/libQt5Widgets.a  C:/Qt/5.15.2/mingw81_64/lib/libQt5Gui.a  C:/Qt/5.15.2/mingw81_64/lib/libQt5Core.a  -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32 && cd ."
CMakeFiles/WGCS.dir/main.cpp.obj:main.cpp:(.text.startup+0x71): undefined reference to `mavsdk::Mavsdk::Mavsdk()'
CMakeFiles/WGCS.dir/main.cpp.obj:main.cpp:(.text.startup+0x8a): undefined reference to `mavsdk::Mavsdk::~Mavsdk()'
CMakeFiles/WGCS.dir/main.cpp.obj:main.cpp:(.text.startup+0xc8): undefined reference to `mavsdk::Mavsdk::~Mavsdk()'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
19:14:19: The process "C:\Qt\Tools\CMake_64\bin\cmake.exe" exited with code 1.
Error while building/deploying project WGCS (kit: Desktop Qt 5.15.2 MinGW 64-bit)
When executing step "CMake Build"
19:14:19: Elapsed time: 00:01.
```

JonasVautherin, 2021-02-11:
I think you don't link MAVSDK correctly. Did you have a look at [how the examples do it](https://github.com/mavlink/MAVSDK/blob/develop/examples/fly_mission/CMakeLists.txt)?

Yogi30, 2021-02-11:
Hi @JonasVautherin, thanks for the reply. Yes, I have checked the examples directory. I'm able to build the example using CMake in windows, but when I tried to do build the same project in Qt creator, I get these errors.

YogeshHC, 2021-02-11:
Hi @JonasVautherin, yes I have checked the examples in the examples directory. I've built and tested them with CMake on windows. But, when I try to build the same example (eg. battery) in Qt Creator(CMake Project) on the same machine I get these errors.

JonasVautherin, 2021-02-11:
But the code you show above is not from the examples, is it? :thinking:

YogeshHC, 2021-02-12:
main.cpp
```
#include <mavsdk/mavsdk.h>
#include <mavsdk/plugins/mavlink_passthrough/mavlink_passthrough.h>
#include <mavsdk/plugins/telemetry/telemetry.h>
#include <chrono>
#include <cstdint>
#include <iostream>
#include <future>
#include <memory>

using namespace mavsdk;

static void subscribe_armed(Telemetry& telemetry);
static void send_battery_status(MavlinkPassthrough& mavlink_passthrough);
static void usage(const std::string& bin_name);

int main(int argc, char** argv)
{
    Mavsdk mavsdk;
    std::string connection_url;
    ConnectionResult connection_result;

    if (argc == 2) {
        connection_url = argv[1];
        connection_result = mavsdk.add_any_connection(connection_url);
    } else {
        usage(argv[0]);
        return 1;
    }

    if (connection_result != ConnectionResult::Success) {
        std::cout << "Connection failed: " << connection_result << std::endl;
        return 1;
    }

    std::promise<void> prom;
    std::future<void> fut = prom.get_future();
    std::cout << "Waiting to discover system..." << std::endl;
    mavsdk.subscribe_on_new_system([&mavsdk, &prom]() {
        const auto system = mavsdk.systems().at(0);

        if (system->is_connected()) {
            prom.set_value();
        }
    });

    if (fut.wait_for(std::chrono::seconds(2)) != std::future_status::ready) {
        std::cout << "No device found, exiting." << std::endl;
        return 1;
    }

    auto system = mavsdk.systems().at(0);

    auto telemetry = Telemetry{system};
    auto mavlink_passthrough = MavlinkPassthrough{system};

    subscribe_armed(telemetry);

    while (true) {
        send_battery_status(mavlink_passthrough);
        std::this_thread::sleep_for(std::chrono::seconds(1));
    }

    return 0;
}

void subscribe_armed(Telemetry& telemetry)
{
    telemetry.subscribe_armed(
[... 96 lines cut ...]
CMakeFiles/WGCS.dir/main.cpp.obj: In function `main':
D:/WGCS/main.cpp:18: undefined reference to `mavsdk::Mavsdk::Mavsdk()'
D:/WGCS/main.cpp:24: undefined reference to `mavsdk::Mavsdk::add_any_connection(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
D:/WGCS/main.cpp:31: undefined reference to `mavsdk::operator<<(std::ostream&, mavsdk::ConnectionResult const&)'
D:/WGCS/main.cpp:38: undefined reference to `mavsdk::Mavsdk::subscribe_on_new_system(std::function<void ()>)'
D:/WGCS/main.cpp:51: undefined reference to `mavsdk::Mavsdk::systems() const'
D:/WGCS/main.cpp:53: undefined reference to `mavsdk::Telemetry::Telemetry(std::shared_ptr<mavsdk::System>)'
D:/WGCS/main.cpp:54: undefined reference to `mavsdk::MavlinkPassthrough::MavlinkPassthrough(std::shared_ptr<mavsdk::System>)'
D:/WGCS/main.cpp:18: undefined reference to `mavsdk::Mavsdk::~Mavsdk()'
D:/WGCS/main.cpp:54: undefined reference to `mavsdk::MavlinkPassthrough::~MavlinkPassthrough()'
D:/WGCS/main.cpp:53: undefined reference to `mavsdk::Telemetry::~Telemetry()'
D:/WGCS/main.cpp:18: undefined reference to `mavsdk::Mavsdk::~Mavsdk()'
CMakeFiles/WGCS.dir/main.cpp.obj: In function `subscribe_armed':
D:/WGCS/main.cpp:68: undefined reference to `mavsdk::Telemetry::subscribe_armed(std::function<void (bool)>)'
CMakeFiles/WGCS.dir/main.cpp.obj: In function `send_battery_status':
D:/WGCS/main.cpp:91: undefined reference to `mavsdk::MavlinkPassthrough::get_our_compid() const'
D:/WGCS/main.cpp:90: undefined reference to `mavsdk::MavlinkPassthrough::get_our_sysid() const'
D:/WGCS/main.cpp:108: undefined reference to `mavsdk::MavlinkPassthrough::send_message(__mavlink_message&)'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
13:14:40: The process "C:\Program Files\CMake\bin\cmake.exe" exited with code 1.
Error while building/deploying project WGCS (kit: Desktop Qt 5.15.2 MinGW 64-bit)
When executing step "CMake Build"
13:14:41: Elapsed time: 00:02.
```

JonasVautherin, 2021-02-12:
Is that a clean build? Does the cmake configure step succeed?

YogeshHC, 2021-02-12:
Yes, it is a clean build.

```
Running C:\Program Files\CMake\bin\cmake.exe -S D:/WGCS -B C:/Users/Admin/AppData/Local/Temp/QtCreator-rIqWBF/qtc-cmake-cNfjpMdL -GNinja "-DCMAKE_BUILD_TYPE:String=Release" "-DQT_QMAKE_EXECUTABLE:STRING=C:/Qt/5.15.2/mingw81_64/bin/qmake.exe" "-DCMAKE_PREFIX_PATH:STRING=C:/Qt/5.15.2/mingw81_64" "-DCMAKE_C_COMPILER:STRING=C:/Qt/Tools/mingw810_64/bin/gcc.exe" "-DCMAKE_CXX_COMPILER:STRING=C:/Qt/Tools/mingw810_64/bin/g++.exe" in C:\Users\Admin\AppData\Local\Temp\QtCreator-rIqWBF\qtc-cmake-cNfjpMdL.
-- The C compiler identification is GNU 8.1.0
-- The CXX compiler identification is GNU 8.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Qt/Tools/mingw810_64/bin/gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Qt/Tools/mingw810_64/bin/g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    QT_QMAKE_EXECUTABLE

-- Build files have been written to: C:/Users/Admin/AppData/Local/Temp/QtCreator-rIqWBF/qtc-cmake-cNfjpMdL
Elapsed time: 00:02.
```

julianoes, 2021-02-12:
> C:/Qt/Tools/mingw810_64/bin/gcc.exe 

It is compiling with Mingw GCC, not MSVC. That's not supported. You can see that there are quite a few places in cmake where we check for `MSVC` to check if it is Windows.

YogeshHC, 2021-02-12:
I missed that. I changed it to `MVSC` and it worked. Than you @julianoes @JonasVautherin

closed 2021-02-12 by YogeshHC

julianoes, 2021-02-12:
Oh nice!