mavlink/MAVSDK #1319: Using multiple UDP remotes

yhabib29, 2021-02-11:
Hello,
I am trying to use MAVSDK to build 2 programs:
- A "client" program which runs on the companion computer on each drone.
- A GroundStation (GS) which can connect to each client.

I am using 2 UDP connections to connect each client to the GS. However, I can create only one UDP remote at a time otherwise it overwrites the existing system (is_single_system in mavsdk_impl.cpp).

How should I connect my GS to the other clients where each program is using Mavsdk.

JonasVautherin, 2021-02-11:
I think you're right: right now MAVSDK can only add one remote. This would need to be updated in the implementation :+1:. Open for contributions, though :innocent:

yhabib29, 2021-02-15:
All right, I will try to implement it in few days.
Thank you for the confirmation 👍🏼

yhabib29, 2021-02-19:
To summarize the UDP connections system in Mavsdk.
- add_udp_connection("IP:PORT"):
    Create a new UdpConnection instance listening on IP:PORT.
    When the first message coming from a new endpoint (IP_ENDPOINT:PORT_ENDPOINT) is received, a new remote is created to send messages to IP_ENDPOINT:PORT_ENDPOINT.
    
- setup_udp_remote(IP, PORT):
    Create a new UdpConnection instance listening on 0.0.0.0:0 (port = 0 ??).
    Then a remote is created to send messages to IP:PORT

When sending a message, the mavsdk_impl iterates over all connections to send the message.
However, if one of the connections is a UDP connection and has no remote, the loop will break even if other connections could send the message.

see [add-multi-udp-remotes](habib29/MAVSDK/tree/add-multi-udp-remotes)

yhabib29, 2021-02-22:
In addition, could it be interesting to add a vector of available system IDs for each connection ?

When sending a message to a specific system ID, this will allow to ignore connections which have never see the corresponding system ID.

The reason is that for radio connections, it might be better to preserve the bandwidth from useless messages.

julianoes, 2021-02-23:
> However, if one of the connections is a UDP connection and has no remote, the loop will break even if other connections could send the message.

That sounds like a bug, right?

> When sending a message to a specific system ID, this will allow to ignore connections which have never see the corresponding system ID.

Yes, I think that would be a good optimization.

yhabib29, 2021-02-23:
> That sounds like a bug, right?

Well, I think yes but I did not know if it was done in purpose ...
Here is the quick changes I did, but there might be a better solution.
``` 
bool MavsdkImpl::send_message(mavlink_message_t& message)
{
    std::lock_guard<std::mutex> lock(_connections_mutex);

    uint succesfull_emissions = 0;
    for (auto it = _connections.begin(); it != _connections.end(); ++it) {
        if ((**it).send_message(message)) {
            succesfull_emissions++;
        }
    }

    if (succesfull_emissions == 0) {
        LogErr() << "send fail";
        return false;
    }

    return true;
}
```

> Yes, I think that would be a good optimization.

All right, I thought of adding a std:vector<uint8_t> in Connection class protected attributes, and a method to check for system ID existence.
I will add it to the corresponding branch, and initialize a pull request.

julianoes, 2021-02-24:
@JonasVautherin can you check this?

julianoes, 2021-03-15:
#1331 was merged, closing.

closed 2021-03-15 by julianoes