mavlink/MAVSDK #1320: [Building Process] macOS 11.2.1 is failing when building mavsdk_server from source

norisg, 2021-02-17:
Related to the following slack conversation: https://px4.slack.com/archives/C52DXKYLV/p1613318568043400

I have created an output file with the error description. 

I guess its a macOS Xcode issue.

I created the output file : https://gist.github.com/norisg/a825d85ea27193ccf588b30f12f100cd

JonasVautherin, 2021-02-17:
Better explain the error here than link to it, for the record. Those links may get unavailable after a while.

The problem is the following: when building MAVSDK in superbuild mode, CMake will build OpenSSL as part of the dependencies. On macOS, it succeeds in configuring but fails to build, with the following error message:

```
**********************************************************************
***                                                                ***
***   OpenSSL has been successfully configured                     ***
***                                                                ***
***   If you encounter a problem while building, please open an    ***
***   issue on GitHub <https://github.com/openssl/openssl/issues>  ***
***   and include the output from the following command:           ***
***                                                                ***
***       perl configdata.pm --dump                                ***
***                                                                ***
***   (If you are new to OpenSSL, you might want to consult the    ***
***   'Troubleshooting' section in the INSTALL file first)         ***
***                                                                ***
**********************************************************************
[ 75%] Performing build step for 'openssl'
../openssl/apps/bf_prefix.c:10:10: fatal error: 'stdio.h' file not found
#include <stdio.h>
         ^~~~~~~~~
../openssl/apps/s_cb.c:11:10: fatal error: 'stdio.h' file not found
#include <stdio.h>
         ^~~~~~~~~
```

JonasVautherin, 2021-02-17:
I think you are missing the `SDKROOT` environment variable to build OpenSSL. It is done [here](https://github.com/mavlink/MAVSDK/blob/develop/.github/workflows/main.yml#L381) in the CI.

Can you try something like below, and run a clean build again?

```
export SDKROOT=$(xcrun --sdk macosx --show-sdk-path)
```

julianoes, 2021-02-17:
This work fine for me:
```
[ 62%] Performing configure step for 'openssl'
Operating system: x86_64-apple-darwinDarwin Kernel Version 20.3.0: Thu Jan 21 00:07:06 PST 2021; root:xnu-7195.81.3~1/RELEASE_X86_64
WARNING! If you wish to build 32-bit libraries, then you have to
         invoke 'KERNEL_BITS=32 /Users/julianoes/src/MAVSDK/build/default/third_party/openssl/openssl/src/openssl/config  '--prefix=/Users/julianoes/src/MAVSDK/build/default/third_party/install' 'no-shared''.
Configuring OpenSSL version 1.1.1h (0x1010108fL) for darwin64-x86_64-cc
Using os-specific seed configuration
Creating configdata.pm
Creating Makefile

**********************************************************************
***                                                                ***
***   OpenSSL has been successfully configured                     ***
***                                                                ***
***   If you encounter a problem while building, please open an    ***
***   issue on GitHub <https://github.com/openssl/openssl/issues>  ***
***   and include the output from the following command:           ***
***                                                                ***
***       perl configdata.pm --dump                                ***
***                                                                ***
***   (If you are new to OpenSSL, you might want to consult the    ***
***   'Troubleshooting' section in the INSTALL file first)         ***
***                                                                ***
**********************************************************************
[ 75%] Performing build step for 'openssl'
ar: creating archive apps/libapps.a
ar: creating archive libssl.a
```

@norisg do you have the command line tools installed?
I think `sudo xcode-select --install` should do it. I did that when `brew` told me to.

julianoes, 2021-02-17:
@norisg next question: are you on M1 or Intel?

norisgOG, 2021-02-18:
Thank you for your time, so I tried the command `export SDKROOT=$(xcrun --sdk macosx --show-sdk-path)` which worked for me, the command `sudo xcode-select --install` was already installed and made no difference.

norisg, 2021-02-18:
So installation worked fine after the ` export SDKROOT=$(xcrun --sdk macosx --show-sdk-path)`

closed 2021-02-18 by norisg