mavlink/MAVSDK #1326: Messages forwarding feature

yhabib29, 2021-02-19:
Hi,

For my projects needs, I had to implement messages forwarding.
An example use case is:
- 245.190 = Ground Control Station (GCS)
- 1.195 = Companion Computer (CC)
- 1.1 = PX4 Autopilot (PX4)

CC is connected to the GCS (UDP) and to the PX4 (serial).
We need the GCS to be able to send command/messages directly to the PX4.

Thus, I have added messages forwarding functionality based on PX4 source code.
The forward_message function was adapted from Firmware v1.11.3:
    https://github.com/PX4/PX4-Autopilot/blob/v1.11.3/src/modules/mavlink/mavlink_main.cpp

See changes in: [add-messages-forwarding](/yhabib29/MAVSDK/tree/add-messages-forwarding)

I had to keep track of the origin connection from which the message was received to ignore it when forwarding.
There might be a better way to do this.

```diff --git a/src/core/connection.h b/src/core/connection.h
index 02ba17b4..5b977325 100644
--- a/src/core/connection.h
+++ b/src/core/connection.h
@@ -8,7 +8,7 @@ namespace mavsdk {
 
 class Connection {
 public:
-    typedef std::function<void(mavlink_message_t& message)> receiver_callback_t;
+    typedef std::function<void(mavlink_message_t& message, Connection *connection)> receiver_callback_t;
 
     Connection(receiver_callback_t receiver_callback);
     virtual ~Connection();
@@ -25,7 +25,7 @@ public:
 protected:
     bool start_mavlink_receiver();
     void stop_mavlink_receiver();
-    void receive_message(mavlink_message_t& message);
+    void receive_message(mavlink_message_t& message, Connection *connection);
 
     receiver_callback_t _receiver_callback{};
     std::unique_ptr<MAVLinkReceiver> _mavlink_receiver;
```

If this feature is useful to you, I can make a pull request.
However, i did not update the integration tests, etc ...

julianoes, 2021-02-23:
I think message forwarding would be a good feature. However, I wouldn't make it an option of the mavsdk configuration, so of the mavsdk instance but rather of connections. So you don't necessarily want forwarding between all connections but maybe only between some of them. So whenever you add a connection, you could specify if it should forward or not.

yhabib29, 2021-02-23:
Oh, good point, that is a better solution !
I will:
- Add a parameter `message_forwarding` in functions creating connection (default to `false`).
- `bool _message_forwarding;` protected attribute in `Connection `class, with a getter public method.
- A check for this boolean in `forward_message `function when iterating over all connections.

Thus, if one connection has the `message_forwarding` option, it will forward messages (after check) received from all other connections.

yhabib29, 2021-03-06:
Resolved in #1330.

closed 2021-03-06 by yhabib29