mavlink/MAVSDK #1327: Connection Error

Guillellorente, 2021-02-23:
Hi all, I'm new here trying to develop some missions in order to be able to do the inspection of a building with a drone. I started following the steps in [https://mavsdk.mavlink.io/develop/en/cpp/quickstart.html](url) but I'm blocked where I have to run the example of the takeoff and landing. I've followed every step precisely and still having the following issue when running the "./takeoff_and_land udp://:14540",  where I'm obtaining the following: 

[06:42:30|Info ] MAVSDK version: v0.36.0 (mavsdk_impl.cpp:21)
Waiting to discover system...
[06:42:30|Info ] New system on: 127.0.0.1:14580 (with sysid: 1) (udp_connection.cpp:190)
[06:42:30|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:418)
[06:42:30|Debug] Component Autopilot (1) added. (system_impl.cpp:324)
[06:42:30|Debug] Discovered 1 component(s) (UUID: 5283920058631409231) (system_impl.cpp:481)
Discovered system
[06:42:30|Debug] MAVLink: info: [logger] ./log/2021-02-23/02_40_40.ulg (system_impl.cpp:230)
[06:42:30|Debug] New: System ID: 255 Comp ID: 190 (mavsdk_impl.cpp:418)
[06:42:30|Debug] Component Unsupported component (190) added. (system_impl.cpp:324)
[06:42:30|Debug] Discovered 1 component(s) (UUID: 255) (system_impl.cpp:481)
Discovered system
Discovered a component with type 0
[06:42:32|Warn ] sending again after 0.50133 s, retries to do: 3  (511). (mavlink_commands.cpp:239)
[06:42:33|Warn ] sending again after 1.00945 s, retries to do: 2  (511). (mavlink_commands.cpp:239)
[06:42:33|Warn ] sending again after 1.51695 s, retries to do: 1  (511). (mavlink_commands.cpp:239)
[06:42:34|Error] Retrying failed (511) (mavlink_commands.cpp:260)
[06:42:34|Warn ] sending again after 0.50922 s, retries to do: 3  (511). (mavlink_commands.cpp:239)
[06:42:35|Warn ] sending again after 1.01631 s, retries to do: 2  (511). (mavlink_commands.cpp:239)
[06:42:35|Warn ] sending again after 1.52332 s, retries to do: 1  (511). (mavlink_commands.cpp:239)
[06:42:36|Error] Retrying failed (511) (mavlink_commands.cpp:260)
Setting rate failed:Connection Error

My guess is that I have to change some mavlinks connection or even the IP, I've tried to change the MAV_BROADCAST to 1 but I'm still having the same issue. I've tried also to reinstall everything and still having the same issue and I did not find anything on the internet about that error. My idea is, after seeing that the examples works, trying to do the same with a fly_mission example and then my target is to create a mission with .csv file where I have all the 3D coordinates and launch it in the SITL (gazebo or jmavsim).

Any ideas where the solution could be?

Thank you all in advance,

julianoes, 2021-02-23:
There have been some changes on the PX4 side that you might run into. Basically, MAVSDK is seeing heartbeats from QGroundControl.

When you're  connecting you should add a check for `system.has_autopilot()`. If not, you keep waiting and ignore it.

julianoes, 2021-02-23:
This should fix it: https://github.com/mavlink/MAVSDK/pull/1328

Can you try it?

Guillellorente, 2021-02-23:
Yes! it worked out perfectly, I will take into account the autopilot check and all the changes made in order to continue with the other examples.  If I'd have future unresolved issues I'll post, I hope will not! 

Thank you very much.

Best regards

julianoes, 2021-02-24:
You're welcome!

closed 2021-02-24 by julianoes