mavlink/MAVSDK #1334: [mavsdk-python SITL] mavsdk-server crashes when reading EulerAngle

norisg, 2021-02-28:
I am using mavsdk-python to control my drone in offboard mode. I had problems when trying to use the EulerAngle and reading them in a cycle around 20 Hz.  After some seconds the mavsdk-server will crash. First I thought it was a problem with the use of global variables in python, but after commenting out the lines which read the EulerAngle, it runs perfectly.

**To Reproduce**
Steps to reproduce the behavior:
1. Switch on SITL(PX4)

2. Start my python program (fcc_bug.py)
3. Start a Testmessage Script which sends messages to the fcc.py
4. See error

norisg, 2021-02-28:
`#!/usr/bin/env python3
import math
import socket
import time
from gpiozero import LED
from os import path
import sys
import serial

import asyncio
from mavsdk import System
from mavsdk.action import ActionError
from mavsdk.telemetry import LandedState,PositionNed,FlightMode
from mavsdk.offboard import (OffboardError, PositionNedYaw, VelocityNedYaw)

#========================================#
#==========Constant Variables============#

# Settings
g_c_maxVV           = 3.0
g_c_minHeight       = 2
g_c_maxHAG          = 115
g_c_tolDTG          = 0.3
g_c_tolAlt          = 0.3
g_c_voltage_0m      = 13.8
g_c_voltage_100m    = 15.8
g_c_maxRadius       = 100
g_c_telemBAUD       = 921600
g_c_telemBAUD       = 57600
g_c_clapOn_s        = 0.5
g_c_clapOff_s       = 0.15
g_c_maxTFHaboveMinH = 0.5
g_c_groundSpeed     = 10

async def run():

    global REQ_FM0_LAND
    global REQ_FM1_HOVER
    global REQ_FM2_RETURN_TO_BASE
    global REQ_FM3_USE_COORDINATES
    global REQ_FM4_START
    global REQ_FM3_ALTITUDE_m
    global REQ_FM3_HEADING_deg
    global REQ_FM3_GEO_m
    global REQ_FM3_GROUND_VELOCITY_mps

    global g_isRPI
    global g_startPoint
    global g_beats
    global g_isInitializationPending
    global g_opTheaterRadius
    global g_tfMode
    global g_connectionSuccess
    global g_velocityMode
    #Definition of Variables:

    #Coordinates are relativ to EKF Origin (THIS IS NOT HOME!!!)
    g_isRPI                     = False
    g_startPoint                = [0, 0, 0, 'New Startpoint']
    g_beats                     = []
    g_isInitializationPending   = True
    g_opTheaterRadius           = 100
    g_tfMode                    = False
    g_connectionSuccess         = False
    g_velocityMode              = False
    #FlightModes:
    REQ_FM0_LAND            = False
    REQ_FM1_HOVER           = False
    REQ_FM2_RETURN_TO_BASE  = False
    REQ_FM3_USE_COORDINATES = False
[... 261 lines cut ...]
            REQ_FM2_RETURN_TO_BASE  = False
            REQ_FM3_USE_COORDINATES = False
            REQ_FM4_START           = False
            isNewCommand            = False

            await asyncio.sleep(0.1)
            #end = time.time()
            #display('Drone Task Process Time: {} ms'.format((end - start)*1000))

       except Exception as e:
             display(e)
             display("=============================")
             display("Error proceed with Emergency Land")
             display("=============================")
             await land(drone)
             streamRX.close()

if __name__ == "__main__":
    try:
        loop = asyncio.get_event_loop()
        loop.run_until_complete(run())

    except KeyboardInterrupt:
        print("Programm aborted, please start again")
`

norisg, 2021-02-28:
The Following is the testmessage.py program which sends dummy data to the fcc.py:

```
import socket
import time

# Communication
# Create a UDP socket
streamTX = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
addressPortTX = ('localhost', 8888)

streamRX = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)

streamRX.bind(('127.0.0.1',8999))

# terminal A
streamRX.setsockopt(socket.SOL_SOCKET, socket.SO_RCVBUF, 512)
# Prints "1024":
print(streamRX.getsockopt(socket.SOL_SOCKET, socket.SO_RCVBUF))
```
```

# Flight Modes
REQ_FM0_LAND            = False
REQ_FM1_HOVER           = False
REQ_FM2_RETURN_TO_BASE  = False
REQ_FM3_USE_COORDINATES = False
REQ_FM4_START           = False
REQ_MESSAGE             = ''

IS_FM0_LAND            = False
IS_FM1_HOVER           = False
IS_FM2_RETURN_TO_BASE  = False
IS_FM3_USE_COORDINATES = False
IS_FM4_START           = False

# Coordinates
REQ_FM3_ALTITUDE_m          = 0
REQ_FM3_HEADING_deg         = 0
REQ_FM3_GEO_m               = [0, 0]
REQ_FM3_GROUND_VELOCITY_mps = 0

IS_FM3_ALTITUDE_m          = 0
IS_FM3_HEADING_deg         = 0
IS_FM3_GEO_m               = [0, 0]
IS_FM3_GROUND_VELOCITY_mps = 0

def SetReq(data):
    global REQ_FM0_LAND
    global REQ_FM1_HOVER
    global REQ_FM2_RETURN_TO_BASE
    global REQ_FM3_USE_COORDINATES
    global REQ_FM4_START
    global REQ_FM3_ALTITUDE_m
    global REQ_FM3_HEADING_deg
    global REQ_FM3_GEO_m
    global REQ_FM3_GROUND_VELOCITY_mps

    REQ_FM0_LAND                = data[0]
    REQ_FM1_HOVER               = data[1]
    REQ_FM2_RETURN_TO_BASE      = data[2]
    REQ_FM3_USE_COORDINATES     = data[3]
    REQ_FM4_START               = data[4]
    REQ_FM3_ALTITUDE_m          = data[5]
    REQ_FM3_HEADING_deg         = data[6]
    REQ_FM3_GEO_m               = [data[7], data[8]]
    REQ_FM3_GROUND_VELOCITY_mps = data[9]

def _send(message = ''):
    message = '{},{},{},{},{},{},{},{},{},{},{}'.format(
[... 103 lines cut ...]
        timeStamps.append(0)
    while len(timeStamps) < len(reqs):
        timeStamps.append(timeStamps[-1] + defaultTimeIncrement)
    print(timeStamps)

    # Continuously sending Data according to timeStamps
    while timeElapsed < timeStamps[len(timeStamps) - 1] + timeLastStamp:
        index = -1
        for i, item in enumerate(timeStamps):
            if timeElapsed >= timeStamps[i] and index + 1 < len(reqs):
                index = index + 1
        if index >= 0:
            req = reqs[index]
            SetReq(req)
            print("Send")
            _send(req[-1])
            print("Receive")
            _receive()
        timeElapsed = time.time() - timeStart
        time.sleep(1)

if __name__ == '__main__':
    main()

```

norisg, 2021-02-28:
This is the Error Log I am getting:

> Waiting for mavsdk_server to be ready...
> Connected to mavsdk_server!
> [main]:Waiting for drone to connect...
> 
> [main]:ConnectionState: [uuid: 0, is_connected: True]
> 
> [main]:Drone discovered with UUID: 0
> 
> [main]:
> ####################################################
> 
> [main]:Mission Start
> 
> Heading: -2.4270238876342773
> Heading: -2.485532522201538
> Heading: -2.582531690597534
> Heading: -2.64835524559021
> Heading: -2.6578786373138428
> Heading: -2.647286891937256
> Heading: -2.6339972019195557
> __anext__ grpc exception
> Traceback (most recent call last):
>   File "///.virtualenvs/mavsdk-python/lib/python3.8/site-packages/aiogrpc/utils.py", line 145, in cb
>     fut.result()
>   File "/usr/local/opt/python@3.8/Frameworks/Python.framework/Versions/3.8/lib/python3.8/concurrent/futures/thread.py", line 57, in run
>     result = self.fn(*self.args, **self.kwargs)
>   File "///.virtualenvs/mavsdk-python/lib/python3.8/site-packages/aiogrpc/utils.py", line 131, in _next
>     return next(self._iterator)
>   File "///.virtualenvs/mavsdk-python/lib/python3.8/site-packages/grpc/_channel.py", line 416, in __next__
>     return self._next()
>   File "///.virtualenvs/mavsdk-python/lib/python3.8/site-packages/grpc/_channel.py", line 786, in _next
>     raise self
> grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
>     status = StatusCode.UNAVAILABLE
>     details = "Connection reset by peer"
>     debug_error_string = "{"created":"@1614542611.595399000","description":"Error received from peer ipv6:[::1]:50051","file":"src/core/lib/surface/call.cc","file_line":1068,"grpc_message":"Connection reset by peer","grpc_status":14}"
> >
> [main]:<_MultiThreadedRendezvous of RPC that terminated with:
>     status = StatusCode.UNAVAILABLE
>     details = "Connection reset by peer"
>     debug_error_string = "{"created":"@1614542611.595399000","description":"Error received from peer ipv6:[::1]:50051","file":"src/core/lib/surface/call.cc","file_line":1068,"grpc_message":"Connection reset by peer","grpc_status":14}"
> >
>

julianoes, 2021-03-01:
@norisg thanks for the issue. What is your mavsdk version from PyPi and which what mavsdk_server version are you using? My suspicion is that something is not matching.

norisg, 2021-03-10:
@julianoes I am using (mavsdk 0.15.0) in the mavsdk and mavsdk_server (v0.36.0-2-g6f71cc15) but it crashes in both ways. When I start mavsdk in the python program with ` drone = System() #mavsdk_server is integrated in this command` or when I run mavsdk_server separately with the command ` #drone = System(mavsdk_server_address='localhost', port=57000)` So one might think that when I start mavsdk server in python it should work

julianoes, 2021-03-26:
@norisg We just released 0.17.0. Could you check if it is fixed now? Otherwise I should look into this.