mavlink/MAVSDK #1360: Error connecting RasPi & Pixhawk 4 Mini using mavsdk_server (running Px4)

beenjamminnallday, 2021-03-17:
Hello,

I'm attempting to have a companion computer (RasPI) on board and my Pixhawk using mavsdk_server. 

```
pi@raspberrypi:~ $ /usr/local/lib/python3.7/dist-packages/mavsdk/bin/mavsdk_server -p 44881 serial:///dev/ttyAMA0
[09:38:48|Info ] MAVSDK version: v0.36.0 (mavsdk_impl.cpp:21)
[09:38:48|Debug] New: System ID: 0 Comp ID: 0 (mavsdk_impl.cpp:418)
[09:38:48|Info ] Server started (grpc_server.cpp:45)
[09:38:48|Info ] Server set to listen on 0.0.0.0:44881 (grpc_server.cpp:46)
[09:38:48|Info ] Waiting to discover system on serial:///dev/ttyAMA0... (connection_initiator.h:21)
[09:38:48|Debug] Component Autopilot (1) added. (system_impl.cpp:324)
[09:38:49|Debug] Discovered 1 component(s) (UUID: 3762846588724065589) (system_impl.cpp:481)
[09:38:49|Info ] System discovered (connection_initiator.h:57)
[09:38:50|Warn ] command denied (512). (mavlink_commands.cpp:166)
[09:38:50|Debug] Falling back to gimbal protocol v1 (mission_impl.cpp:131)
[09:38:50|Debug] Falling back to Gimbal Version 1 (gimbal_impl.cpp:58)
[09:38:50|Warn ] sending again, retries to do: 3  (SYS_FAILURE_EN). (mavlink_parameters.cpp:566)
[09:38:51|Warn ] command denied (512). (mavlink_commands.cpp:166)
[09:38:51|Warn ] sending again, retries to do: 2  (SYS_FAILURE_EN). (mavlink_parameters.cpp:566)
[09:38:52|Warn ] sending again, retries to do: 1  (SYS_FAILURE_EN). (mavlink_parameters.cpp:566)
[09:38:53|Error] Error: Retrying failed get param busy timeout: SYS_FAILURE_EN (mavlink_parameters.cpp:582)
[09:38:55|Warn ] sending again, retries to do: 3  (NAV_FT_DST). (mavlink_parameters.cpp:566)
[09:38:56|Warn ] sending again, retries to do: 3  (NAV_FT_RS). (mavlink_parameters.cpp:566)
[09:38:50|Warn ] command denied (512). (mavlink_commands.cpp:166)
```
I am not entirely sure what is causing this issue, I flashed my pixhawk with PX4 so its not similar to issue #1134 

Any info would be very helpful.

JonasVautherin, 2021-03-17:
After that, does it work, or not? Like can you get telemetry and send actions?

beenjamminnallday, 2021-03-17:
My mistake after running some of the example scripts it works. I assumed it wouldnt due to the error messages when I started the server.

closed 2021-03-17 by beenjamminnallday

JonasVautherin, 2021-03-17:
No worries.

An interesting thing to know is that "command denied" means that the command was sent by MAVSDK, reached PX4, which answered "DENIED", and that response reached MAVSDK back. So that's not a communication problem, it's more that PX4 was not ready for that command at that point, for some reason.