mavlink/MAVSDK #1369: Windows release build not building

julianoes, 2021-03-22:
I just tried to build in Release mode on Windows with MSVC 2019 and it failed with:

TODO add output:
```

```

julianoes, 2021-03-22:
I think the underlying problem is that the dependencies are all built in debug mode for Windows because MSVC based generators require `--config Release` unlike Makefile based generators which work with `CMAKE_BUILD_TYPE`.

JonasVautherin, 2021-03-22:
So this line does not work properly? https://github.com/mavlink/MAVSDK/blob/main/third_party/cmake/build_target.cmake#L37

JonasVautherin, 2021-03-22:
Oh, this line does not run the dependencies. Got it!

julianoes, 2021-03-22:
While I look at this I'm asking myself: Does it work for Linux because I don't see that we pass on `CMAKE_BUILD_TYPE`?

> ```BUILD_COMMAND cmake --build . --config ${CMAKE_BUILD_TYPE}```

This would only work if `CMAKE_BUILD_TYPE` is set but I assume it's not on Windows where we use `--config`.

JonasVautherin, 2021-03-22:
For the "main" [build command](https://github.com/mavlink/MAVSDK/blob/main/third_party/cmake/build_target.cmake#L37), we only do it for MSVC, where we set `CMAKE_BUILD_TYPE` [in the CI](https://github.com/mavlink/MAVSDK/blob/main/.github/workflows/main.yml#L509).

For the dependencies, maybe we would need to `if (MSVC)`, but in my experience (just tried once on my laptop :laughing:), Linux seems to ignore `--config` completely :see_no_evil:

julianoes, 2021-03-22:
Note that `CMAKE_BUILD_TYPE` does not do anything when you use it with the MSVC generator, as we do it. It just builds debug. You would need `--config Release` as well.

> Linux seems to ignore --config completely 🙈

Yes. When you use Makefile base generators, it is not used. I believe configuration is a build time flag whereas `CMAKE_BUILD_TYPE` is a configure time flag.

JonasVautherin, 2021-03-22:
> Note that CMAKE_BUILD_TYPE does not do anything when you use it with the MSVC generator, as we do it. It just builds debug. You would need --config Release as well.

Yes, but that's exactly what the following would do, right? Be ignored by Linux, and set `CMAKE_BUILD_TYPE` for Windows. And if not specified, either default to Debug, or fail on Windows.

```
BUILD_COMMAND cmake --build . --config ${CMAKE_BUILD_TYPE}
```

julianoes, 2021-03-22:
Yes, true, so what we need to do for Windows is use `--config` as well as `-DCMAKE_BUILD_TYPE`, right?

julianoes, 2021-03-22:
Ok, I can build release on Windows by combining both flags:
```
cmake -DCMAKE_BUILD_TYPE=Release --config Release -Bbuild -H.
```

I think the error of this issue is unrelated to the build type though.

JonasVautherin, 2021-03-22:
Wait, this is the configure step, right? `--config Release` should go in the build step, i.e. `cmake --build build --config Release`, right?

julianoes, 2021-03-22:
You're right. So you need to have `-DCMAKE_BUILD_TYPE` for the configure step and `--config` for the build step.

PR #1371 (2021-03-22): cmake: fix MSVC release build with shared libs
Fixes #1369.
closed 2021-03-22 by JonasVautherin