mavlink/MAVSDK #1383: Build failure on macOS 10.14 (Mojave)

carlocab, 2021-03-25:
Building MAVSDK fails on macOS 10.14 (using the Command Line Tools SDK) with the error message

```
error: 'path' is unavailable: introduced in macOS 10.15
```

Reproducing this error requires applying the patch in #1382, or else the build will fail with

```
/tmp/mavsdk-20210325-1418-1sgyr7x/src/core/tcp_connection.cpp:164:18: error: use of undeclared identifier 'MSG_NOSIGNAL'
    auto flags = MSG_NOSIGNAL;
                 ^
1 error generated.
```

Can the build failure be fixed, or do you no longer support macOS 10.14?

Related: https://github.com/Homebrew/homebrew-core/pull/73923

JonasVautherin, 2021-03-25:
How bad is it if we no longer support Mojave? I guess you'll have to update your system at some point, right? 😇

carlocab, 2021-03-25:
That's up to you, really. I don't mind either way. 

I'm just trying to make sure that Homebrew isn't mistakenly [requiring a macOS version newer than Mojave](https://github.com/Homebrew/homebrew-core/pull/73923/commits/08bbe53e82ddc3248a969733498bdfb80e144f99) to install MAVSDK.

In fact, if you say you don't support Mojave, then that's less work for me :)

JonasVautherin, 2021-03-25:
Ok! Let's "drop" support for Mojave, then! Thanks a lot for reporting!

closed 2021-03-25 by JonasVautherin

carlocab, 2021-05-19:
As it turns out, all that was needed was to build MAVSDK with a newer toolchain than the one provided by the system. (Though I guess that shouldn't be surprising.)

Just thought to let you know in case it's useful elsewhere (or for whomever else stumbles upon this issue).