mavlink/MAVSDK #1386: How does "Pause action" work during mission

doman93, 2021-03-26:
Hello,  I am studying now the interrupt mechanism behind mission pause in the example of mavsdk

https://github.com/mavlink/MAVSDK/blob/main/examples/fly_mission/fly_mission.cpp

In this example, the variable "want_to_pause" is an atomoic bool variable and initialized to be false (LINE 202).
Then author subscribe to mission process and set it true if the mission_progress.current  >=2 to activate the interruption.
This interruption will start the code to pause the mission, then wait for 5s and then resume the mission.

However, i have some questions:
-  I do not observer the variable "want_to_pause" is reset to "false"; thereby this interruption will occur only one at mission_progress.current  ==2 , is it right?
- During the pause process, I can use the offboard mode ?
-  The code L228-230  concerning " while  (!want_to_pause)" serves for which reason?

julianoes, 2021-03-29:
> I do not observer the variable "want_to_pause" be reset to "false"; thereby this interruption will occur only once at mission_progress.current ==2, i.e. mission >=3 will not call this interruption , is it right? How do I enable the pause properly, change "want_to_pause" variable back to "false" at line L263?

It's only set to `true` once, correct.

You could also call it at every waypoint like this:

```
    mission.subscribe_mission_progress([&want_to_pause](Mission::MissionProgress mission_progress) {
        want_to_pause = true;
    });
```

Then you would have to reset it somewhere else, e.g.
```
while (!want_to_pause) {
    sleep_for(seconds(1));
}
want_to_pause = false;
```

> During the pause process, I can use the offboard mode ?

Yes you can. Pause just means it goes to Hold/Loiter mode.

> The code L228-230 concerning " while (!want_to_pause)" serves for which reason? It has any connection with the interruption ?

This waits until the flag is set to `true`.

doman93, 2021-03-29:
Thanks @julianoes

closed 2021-03-29 by doman93