mavlink/MAVSDK #1387: Compiling error on Raspberry Pi 4

cjffly, 2021-03-30:
Hi, while I am trying to compile the latest mavsdk (0.38.2) on my rPi 4, I had this error when I try to install:
```
mavsdk/src/core/mavlink_parameters.cpp:279:23: error: too few arguments to function ‘uint16_t mavlink_msg_param_ext_request_read_pack(uint8_t, uint8_t, mavlink_message_t*, uint8_t, uint8_t, const char*, int16_t, uint8_t)’
  279 |                     -1);
      |                       ^

At global scope:
cc1plus: note: unrecognized command-line option ‘-Wno-unknown-warning-option’ may have been intended to silence earlier diagnostics
make[2]: *** [src/core/CMakeFiles/mavsdk.dir/build.make:252: src/core/CMakeFiles/mavsdk.dir/mavlink_parameters.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:752: src/core/CMakeFiles/mavsdk.dir/all] Error 2
make: *** [Makefile:150: all] Error 2
```

I thought my gcc version was 8 so I took 4 hours to get the latest gcc version (10.2.0), and this error is still there. Any idea what cause this error?

JonasVautherin, 2021-03-30:
Sounds like you use `mavsdk` as a git submodule, and you updated it without updating its submodules (i.e. mavlink). Could it be?

julianoes, 2021-03-30:
@cjffly try `git submodule update --init --recursive` in MAVSDK.

JonasVautherin, 2021-03-30:
Can you show the cmake commands you use to configure and build?

I wonder why it fails at build time, I would have expected it to fail at configure time, [here](https://github.com/mavlink/MAVSDK/blob/main/src/CMakeLists.txt#L36) :thinking:

cjffly, 2021-03-30:
I am using the standard configure and build commands:
```
cmake -DCMAKE_BUILD_TYPE=Debug -DBUILD_SHARED_LIBS=ON -Bbuild/default -H.
cmake --build build/default
```

Here is the part from configure output which contains tinyxml2:
```
-- Preparing external project "tinyxml2" with args:
-- -- -DCMAKE_INSTALL_PREFIX:PATH=......mavsdk/build/default/third_party/install
-- -- -DCMAKE_TOOLCHAIN_FILE:PATH=
-- -- -DCMAKE_BUILD_TYPE=Debug
-- -- -DCMAKE_POSITION_INDEPENDENT_CODE=ON
-- -- -DBUILD_SHARED_LIBS=OFF
-- -- -DBUILD_TESTS=OFF
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    DEPLOYMENT_TARGET
    ENABLE_STRICT_TRY_COMPILE
    PLATFORM

-- Build files have been written to: ......mavsdk/build/default/third_party/tinyxml2
Scanning dependencies of target tinyxml2
[ 12%] Creating directories for 'tinyxml2'
[ 25%] Performing download step (download, verify and extract) for 'tinyxml2'
-- Downloading...
   dst='......mavsdk/build/default/third_party/tinyxml2/tinyxml2/src/7.1.0.tar.gz'
   timeout='none'
-- Using src='https://github.com/leethomason/tinyxml2/archive/7.1.0.tar.gz'
-- [download 100% complete]
-- Downloading... done
-- extracting...
     src='......mavsdk/build/default/third_party/tinyxml2/tinyxml2/src/7.1.0.tar.gz'
     dst='......mavsdk/build/default/third_party/tinyxml2/tinyxml2/src/tinyxml2'
-- extracting... [tar xfz]
-- extracting... [analysis]
-- extracting... [rename]
-- extracting... [clean up]
-- extracting... done
[ 50%] No update step for 'tinyxml2'
[ 50%] No patch step for 'tinyxml2'
[ 62%] Performing configure step for 'tinyxml2'
CMake Deprecation Warning at CMakeLists.txt:11 (cmake_policy):
  The OLD behavior for policy CMP0063 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: ......mavsdk/build/default/third_party/tinyxml2/tinyxml2/src/tinyxml2-build
[ 75%] Performing build step for 'tinyxml2'
Scanning dependencies of target tinyxml2
[ 50%] Building CXX object CMakeFiles/tinyxml2.dir/tinyxml2.cpp.o
[100%] Linking CXX static library libtinyxml2d.a
[100%] Built target tinyxml2
[ 87%] Performing install step for 'tinyxml2'
[100%] Built target tinyxml2
Install the project...
-- Install configuration: "Debug"
-- Installing: ......mavsdk/build/default/third_party/install/lib/libtinyxml2d.a
-- Installing: ......mavsdk/build/default/third_party/install/include/tinyxml2.h
-- Installing: ......mavsdk/build/default/third_party/install/lib/pkgconfig/tinyxml2.pc
-- Installing: ......mavsdk/build/default/third_party/install/lib/cmake/tinyxml2/tinyxml2Config.cmake
-- Installing: ......mavsdk/build/default/third_party/install/lib/cmake/tinyxml2/tinyxml2ConfigVersion.cmake
-- Installing: ......mavsdk/build/default/third_party/install/lib/cmake/tinyxml2/tinyxml2Targets.cmake
-- Installing: ......mavsdk/build/default/third_party/install/lib/cmake/tinyxml2/tinyxml2Targets-debug.cmake
[100%] Completed 'tinyxml2'
[100%] Built target tinyxml2
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
```

cjffly, 2021-03-30:
> 
> Can you show the cmake commands you use to configure and build?
> 
> I wonder why it fails at build time, I would have expected it to fail at configure time, [here](https://github.com/mavlink/MAVSDK/blob/main/src/CMakeLists.txt#L36) 🤔

OK, seems I figure out that I add my `include($ENV{VCPKG_ROOT}/scripts/buildsystems/vcpkg.cmake)` at the wrong CMakeLists.txt. Previous I thought I should add this into the one under MAVSDK root folder. Now I add the "include vcpkg" to MAVSDK/src/CMakeLists.txt, and build is success.

However, if I don't have vcpkg this will be a problem, so I guess it still needs some kind of fix.

cjffly, 2021-03-31:
@JonasVautherin 
When you mean 

> Can you show the cmake output related to [this line](https://github.com/mavlink/MAVSDK/blob/main/src/CMakeLists.txt#L36)?

Which command output are you referring to? Is it `cmake -DCMAKE_BUILD_TYPE=Debug -DBUILD_SHARED_LIBS=ON -Bbuild/default -H.` or `cmake --build build/default`? If the first, I did posted above, that is the output related to tinyxml2, I did see it download and builds for me, it just couldn't find it.

By the way, I am using raspberry pi 4, not Windows, not sure 

> Windows calls it `libtinyxml2d.a` instead of `libtinyxml2.a`. Maybe there is something to do in the [find module](https://github.com/mavlink/MAVSDK/blob/main/src/cmake/Modules/Findtinyxml2.cmake).

is true. But I will try in release build and let you know the result.

JonasVautherin, 2021-03-31:
Right, sorry I got confused, it's now Windows :see_no_evil:.

So what has changed recently is that we used to build all the dependencies in Release mode, and now we propagate the `CMAKE_BUILD_TYPE` to them. And tinyxml2 does this thing of renaming the debug output file...

> Which command output are you referring to?

I'm looking for some output [like this](https://github.com/mavlink/MAVSDK/blob/main/src/cmake/Modules/Findtinyxml2.cmake#L5)? I.e. when `find_package` or `pkgconfig` finds the library, usually it outputs something like "found tinyxml2 here: <path>".

Maybe the find module should look for tinyxml2d somewhere if `CMAKE_BUILD_TYPE=Debug`... :thinking:. Would you mind having a look at that?

julianoes, 2024-05-27:
Old, closing.

closed 2024-05-27 by julianoes