mavlink/MAVSDK #1391: Structure Cmake for Cpp package managers

bazfp, 2021-03-31:
It would be useful to be able to use MAVSDK with cpp package managers (Hunter, Conan etc) for automatic dependency resolution. Instead of depending on a `SUPERBUILD` or local dependencies.

I've been implementing this library with [hunter](https://github.com/cpp-pm/hunter) and have found success by doing the following:

0. `SUPERBUILD=OFF`
1. Remove git commands from Cmakelists.txt, such as getting version from git tags
2. Don't use git submodules 
3. Remove pkgconfig authoritative behaviour https://github.com/mavlink/MAVSDK/blob/main/src/cmake/Modules/Findtinyxml2.cmake#L25

This won't work for `mavsdk_server` but it is a start. There is likely to be more changes to this to get it to work (for Hunter I have included Hunter_gate but obviously this differs for each manager :) )

JonasVautherin, 2021-03-31:
How did you update `Findtinyxml2.cmake`? Does it still work for the other cases?

bazfp, 2021-03-31:
I removed lines 18-23.

It is unlikely to work in other cases - that why I have not submitted a PR.

JonasVautherin, 2021-03-31:
But why doesn't `find_package(TINYXML2)` work in the first place? Isn't the package manager (e.g. hunter) supposed to work with it? Like by setting some `CMAKE_PREFIX_PATH` or something?

We use pkgconfig as a fallback already...

bazfp, 2021-04-01:
Yes you're correct. It was not working due to the aliased lib names in the provided Find cmake modules. I no longer use the provided `src/cmake/Modules/Find*` modules now and lean on the package manager for finding packages. I have had to rename the libs that were linked as they were incorrect.

`Jsoncpp::jsoncpp` -> `jsoncpp_lib`
`tinyxml2::tinyxml2` -> `tinyxml2`

JonasVautherin, 2021-05-05:
FYI, I believe it should be fixed for jsoncpp using https://github.com/mavlink/MAVSDK/pull/1395. It's not a release yet, so I don't really know how it goes with Hunter, but I think jsoncpp now has the behavior you expect.

bazfp, 2021-05-14:
Aha, Good news :)

I will wait for fresh `jsoncpp` & `tinyxml2` releases with the namespaced library names, and push them to hunter.

Once this is all done I will update the PR #1393, it should be a relatively light-touch hunter integration at that point :)

JonasVautherin, 2021-05-17:
@bazfp: Updated tinyxml2 here: https://github.com/mavlink/MAVSDK/pull/1436

julianoes, 2024-05-27:
@bazfp added hunter support.

closed 2024-05-27 by julianoes