mavlink/MAVSDK #1397: Link libc statically for Android

JonasVautherin, 2021-04-06:


JonasVautherin, 2021-04-06:
@Katawann: I think that you may be right in that MAVSDK-Java should not bring `libc++_shared.so` with it, and that's the reason for your issue. I do believe that the solution would be to build `libmavsdk_server.so` as a static library. We already do that for `mavsdk_server` (that we build as a static executable), so that should be similar (where instead of building a static executable `mavsdk_server`, for Android we want `libmavsdk_server.so` to link everything statically).

If you look e.g. [here](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk_server/src/CMakeLists.txt#L102-L107), you'll find this `BUILD_STATIC_MAVSDK_SERVER` that we use in our CMakeLists.txt.

But [this is not done for Android or iOS](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk_server/src/CMakeLists.txt#L92), where we only build `libmavsdk_server.so`. Maybe you could try to make it such that for Android, it does link statically into `libmavsdk_server.so`.

Does that make sense?

JonasVautherin, 2022-05-24:
Closing due to inactivity. Feel free to reopen if necessary.

closed 2022-05-24 by JonasVautherin

MustafaMunir, 2022-11-24:
@JonasVautherin , I am facing the issue discussed https://github.com/mavlink/MAVSDK-Java/issues/65 while developing an application which requires NDK 21. I feel that the workaround for me as well is to build mavsdk_server as static library. Can you please guide me if it require changes to build or has already been done. Thanks a lot.