mavlink/MAVSDK #1410: What problems will we encounter by using Ardupilot + MAVSDK ?

mommotti, 2021-04-12:
My team is willing to employ Ardupilot over PX4, since the drone we are using comes with Ardupilot by default...
We acknowledge this thread https://github.com/mavlink/MAVSDK/issues/728
but we are still wondering the future effects of using Ardupilot. ( takeoff_and_land works for us! )

julianoes, 2021-04-12:
Two important things that are not (yet) compatible:
- mission upload/download (AP: 0 item is home, PX4: 0 item is first item)
- flight modes

mommotti, 2021-04-12:
> 
> 
> Two important things that are not (yet) compatible:
> 
>     * mission upload/download (AP: 0 item is home, PX4: 0 item is first item)
> 
>     * flight modes

Thank you again Julian! We can call this issue closed.
Is there a Discord server for MAVSDK?

JonasVautherin, 2021-04-12:
One thing that would be cool would be to have some easy way to run SITL with Ardupilot. I'm sure it exists, but I am not very familiar with Ardupilot. For instance with PX4, there is [px4-gazebo-headless](https://github.com/jonasvautherin/px4-gazebo-headless) that is just a headless docker container running PX4 and gazebo.

Having an equivalent for Ardupilot would be a first step towards adding integration tests for it, and slowly modify the code where needed to support woth Ardupilot and PX4 :+1:.

> Is there a Discord server for MAVSDK?

There is the PX4 Slack, channel #MAVSDK: http://slack.px4.io/ :blush:

closed 2021-04-12 by julianoes