mavlink/MAVSDK #1411: How to connect a drone using system id with mavsdk

Kutbip, 2021-04-14:
Hello Everyone,
I am using MAVSDK-Java to monitoring drones.
Currently, MAVSDK java only exposes to connect drones using UDP or TCP which is working perfectly.
Now, to connect with multiple drones, the app requires connecting using the system id which I am not able to do.

Later I came to know that the C++ library can do the task.
I have tried to build a C++ library  [here](https://github.com/mavlink/MAVSDK.git) to use with Android, but I can't make a build due to some errors. followed as per the documentation.

So the questions are 
1. Can I achieve the above(connect the drone with system id) using C++ lib.
2. Can I get the .so file of this C++ lib so that I can use it in android through JNI.

FYR: @JonasVautherin @julianoes

JonasVautherin, 2021-04-14:
What about using MAVSDK-Java, but running multiple instances of `MavsdkServer`? Say one that listens on UDP:14540, and one that listens on UDP:14541. Then your two drones should broadcast on 14540 and 14541 respectively.

I am personally not a big fan of having all the drones connecting to the app on 14540, to be honest. IMO, on the app side, it should be "one drone, one socket", and my suggestion above does exactly that :+1:.

Kutbip, 2021-04-14:
Thanks for the quick reply @JonasVautherin.
I understood what you explained however I have to connect the drone using a system id. (project requirement)
Is there any way with mavsdk java?
I am ready to use C++ in android but somehow I couldn't able to build it.

JonasVautherin, 2021-04-14:
> Is there any way with mavsdk java?

No, there isn't. We would need to discuss how we would want to expose that. I know that not everybody has the same opinion as I do, but I would push for doing what I suggested above, and if there was the need for a discovery protocol, I would do that separately, and not by sending everything on the same UDP port on the client :see_no_evil:.

Kutbip, 2021-04-15:
We have used the same process that u suggest with simulate environment.
We have used UDP port to communicate a drone with Gazebo simulator.
We have created a mission from QGC and monitoring on our android app and controlling the mission like pause, resume and terminate.
But
For real environment or in production, we have to connect a drone with system id to monitoring active mission.
In our application, we have to register drones and we will get a system id while registration.
So we have to connect using that system id.
@JonasVautherin

JonasVautherin, 2021-04-15:
Do you have a fixed number of drones in the real environment, or do you have "n drones" and you don't know how many?

Kutbip, 2021-04-15:
Not fixed number of drones but we have max 5-6 drones.

Kutbip, 2021-04-15:
Hello @JonasVautherin,
Actually, we are new to this development so bit curious about how drone communication will work in a real environment.
Till now we have used the simulated environment as I mentioned using the Gazebo simulator.

JonasVautherin, 2021-04-15:
> Not a fixed number of drones but we have a max of 5-6 drones.

The solutions I see (none of which is straightforward):

1. Using MAVSDK-Java: implement a discovery protocol, have your app discover the drones and their UDP port, and start one `MavsdkServer` instance for each drone
2. Using MAVSDK-Java: let the user enter the UDP ports for each drone manually and create a `MavsdkServer` instance for each
3. Use MAVSDK-C++ and its multi-systems feature

Kutbip, 2021-04-25:
Hello @JonasVautherin,
Thanks for the solutions. We have considered #3.
I have successfully created a C++ build (.so file) for Android to use.
Could you please provide any reference that can help me to use C++ MAV SDK classes to use in Android.?
Which class or functions should I use to connect drones with system id and set the actions.
Appriciate your help..!!!!

closed 2024-05-27 by julianoes