mavlink/MAVSDK #1413: set_rate missing for rawGps

cjffly, 2021-04-15:
Hi,

rawGps info now can be accessed with v0.38+, but I found that its missing something like "telemetry.set_rate_rawGps()" as the position can do "telemetry.set_rate_position()". Is this a missing feature or there is some other way to set the rawGps access rate?

Best

julianoes, 2021-04-16:
Hi, good question. You can actually just use `set_rate_gps_info` as the underlying mavlink message is the same.

@JonasVautherin should we add it nevertheless or add a comment to clarify this?

JonasVautherin, 2021-04-16:
I first was thinking about adding it, and handling the case where somebody does `set_rate_gps_info()` AND `set_rate_raw_gps` like this:

* If both `subscribe_gps_info` and `subscribe_raw_gps` are active, use the max of both rate values
* If only one is active, take its rate

The problem is for the polling interface: what if I want to poll gps_info very fast, but subscribed to raw_gps at a slow rate? Or maybe something like:

* If both `subscribe_gps_info` and `subscribe_raw_gps` are active, use the max of both rate values
* If only one is active, take its rate
* If `set_rate_` has been called (even if the subscription is inactive), use that rate in the max computation

Which then means that if you do `set_rate_gps_info` and then only poll `raw_gps`, you see the effect of the new rate (if that rate was faster than the default). Not sure if that's a problem, though... And we should document the behavior, and maybe add a log line when `set_rate_` is being set to the lower value and is therefore ignored.

Not perfect, but that's the best I can think of to have a rather natural way of using both rates :sweat_smile: