mavlink/MAVSDK #1428: Feature req: pkgconfig for building

vranki, 2021-05-06:
The "standard" way of including libraries in linux is pkgconfig mechanism. This tells what headers and libraries must be included in the compilation process. Almost any library in linux distros ship with pkgconfig metadata file. Pkgconfig can be used from any build system including cmake and qmake.

Mavsdk currently does not ship with pkgconfig and it's not obvious how to build correctly when using other than CMake as build system.

julianoes, 2021-05-06:
Thanks for the note. Pull requests are welcome :wink:.

JonasVautherin, 2021-05-06:
Yes, that would be a great addition! Would you be willing to look into it?

vranki, 2021-05-06:
I can take a look.

First thing I noticed that the debian package doesn't provide library symlinks. For example I needed to ln -s /usr/lib/libmavsdk.so.0 /usr/lib/libmavsdk.so

I think that's a packaging error.

After that this .pc file placed at /usr/lib/x86_64-linux-gnu/pkgconfig/mavsdk.pc seems to work or at least compile:

```
#############################################
#####   Pkg-Config file for mavsdk      #####
#############################################

prefix=/usr
exec_prefix=${prefix}

libdir=/usr/lib
includedir=${prefix}/include/mavsdk

Name: mavsdk
Version: 0.39.0
Description: Set of libraries providing a high-level API to MAVLink
Libs: -L${libdir} -lmavsdk
Cflags: -I${includedir}
```

JonasVautherin, 2021-05-06:
Should the `.pc` file be generated by cmake? Or only installed? Like can we get the `Libs` and `Cflags` set automatically by cmake? Would be cool :sunglasses:

vranki, 2021-05-10:
I haven't actually done any of them in real life situation, but the debian/rpm package should provide the .pc file. The file contents can be generated by packaging and include for example correct version number. I believe static file without version would also work, but you couldn't check for certain version in project build.

I bet there is some cmake based library in wild with proper .pc packaging that could be used as a example.

JonasVautherin, 2021-05-19:
@vranki just realized it is done [here](https://github.com/leethomason/tinyxml2/blob/master/CMakeLists.txt#L123-L129), which could serve as inspiration:

```cmake
## pkg-config

configure_file(cmake/tinyxml2.pc.in tinyxml2.pc.gen @ONLY)
file(GENERATE OUTPUT tinyxml2.pc INPUT "${CMAKE_CURRENT_BINARY_DIR}/tinyxml2.pc.gen")
install(
    FILES "${CMAKE_CURRENT_BINARY_DIR}/tinyxml2.pc"
    DESTINATION "${tinyxml2_INSTALL_PKGCONFIGDIR}"
    COMPONENT tinyxml2_development
)
```

Using the following `tinyxml2.pc.in`:

```pkgconfig
prefix=@CMAKE_INSTALL_PREFIX@
exec_prefix=${prefix}
libdir=${exec_prefix}/@CMAKE_INSTALL_LIBDIR@
includedir=${prefix}/@CMAKE_INSTALL_INCLUDEDIR@

Name: TinyXML2
Description: simple, small, C++ XML parser
Version: @tinyxml2_VERSION@
Libs: -L${libdir} -l$<TARGET_FILE_BASE_NAME:tinyxml2::tinyxml2>
Cflags: -I${includedir}
```

Consti10, 2022-06-09:
any updates on this ? I am trying to integrate mavsdk with qmake (Qt), which doesn't speak cmake.

Consti10, 2022-06-09:
Dirty but compiles in qmake:
```
# mavsdk - dirty
INCLUDEPATH += /usr/local/include/mavsdk
LIBS += -L/usr/local/lib -lmavsdk
```

JonasVautherin, 2022-06-10:
Did you try with @vranki's mavsdk.pc file above?

JonasVautherin, 2022-08-24:
I started looking into it [(draft PR here)](https://github.com/mavlink/MAVSDK/pull/1889). Happy to get feedback :+1:

julianoes, 2022-09-06:
@vranki could you please give feedback on the draft PR by @JonasVautherin in https://github.com/mavlink/MAVSDK/pull/1889?

PR #1889 (2022-09-06): Add support for pkgconfig
This adds support for pkgconfig. Closes #1428.

The way to get the library in cmake using pkgconfig is:

```
find_package(PkgConfig REQUIRED)
pkg_check_modules(MAVSDK REQUIRED mavsdk)

target_link_libraries(gimbal
    ${MAVSDK_LIBRARIES}
)

target_include_directories(gimbal
    PRIVATE ${MAVSDK_INCLUDE_DIRS}
)
```
closed 2022-09-06 by julianoes