mavlink/MAVSDK #1440: "set_actuator" example doesn't work well, Warning "command unsupported(187)"

YANG-DEMIN, 2021-05-22:
Target: I want to use MAIN 5 outputing the pwm value I set in Quadrotor.
MAVSDK version: `0.40.0 `
Code: the newest code (2021.5.22), `MAVSDK/example/set_actuator`
Simulator: `PX4(V1.10.0) + Gazebo(Quadrotor)`

**### Erro:**
    At first, when i run the other programs, they work well. such as takeoff_land, and fly_mission.
    But when i run the set_actuator, `./set_actuator udp://:14540 1 0.5`, I got the wrong, as follows:
```
[10:02:21|Info ] MAVSDK version: v0.40.0 (mavsdk_impl.cpp:21)
Waiting to discover system...
[10:02:21|Info ] New system on: 127.0.0.1:14580 (with sysid: 1) (udp_connection.cpp:191)
[10:02:21|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:538)
[10:02:21|Debug] Component Autopilot (1) added. (system_impl.cpp:354)
[10:02:22|Debug] Setting UUID to: 5283920058631409231 (system_impl.cpp:222)
[10:02:22|Debug] Discovered 1 component(s) (UUID: 5283920058631409231) (system_impl.cpp:511)
Discovered autopilot
Setting actuator...
[10:02:22|Warn ] command unsupported (187). (mavlink_commands.cpp:173)
Setting actuator failed:Unknown
```
Then, the code is finished. 

I tried some methods to handle it, such as 
1. adding output in the mixer of sitl aircraft.
```
M: 1
O:      10000  10000      0 -10000  10000
S: 0 5  10000  10000      0 -10000  10000
```
2. Armed the vehicle at first, then set actautor. It's useless.

The answer I want to know is that 
1. the `index `variables (1- 6)  in` action.set_actuator(index, value),` what meaning it is,` control group` or anything?
2. How to run the code successfully?

Thank you very much!
Yang

JonasVautherin, 2021-05-22:
> [10:02:22|Warn ] command unsupported (187). (mavlink_commands.cpp:173)

This means that PX4 received the command and responded that it is unsupported ([see here](https://github.com/mavlink/MAVSDK/blob/main/src/core/mavlink_commands.cpp#L172-L173)). Which version of PX4 do you have? @coderkalyan which version of PX4 is required?

YANG-DEMIN, 2021-05-22:
> This means that PX4 received the command and responded that it is unsupported ([see here](https://github.com/mavlink/MAVSDK/blob/main/src/core/mavlink_commands.cpp#L172-L173)). Which version of PX4 [...]

yes, I know where the warning comes from, but how can I fix it?
Thanks very much!

JonasVautherin, 2021-05-22:
Well if your PX4 says that it is not supported, it sounds like your version of PX4 is too old for this feature. That's why I asked which version you have :slightly_smiling_face:

coderkalyan, 2021-05-23:
> @coderkalyan which version of PX4 is required?

@JonasVautherin @YANG-DEMIN Ah, unfortunately this feature is really new - you need to be running PX4 beta to utilize it. It will be supported once the PX4 v1.12 stable is released (hopefully quite soon).

YANG-DEMIN, 2021-05-24:
> > @coderkalyan which version of PX4 is required?
> 
> @JonasVautherin @YANG-DEMIN Ah, unfortunately this feature is really new - you need to be running PX4 beta to utilize it. It will be supported once the PX4 v1.12 stable is released (hopefully quite soon).

Thanks very much! I am looking forward the v1.12! 
As you said, I tried the PX4 beta and it rebacks no erro. But I am still  not sure how to utilize it. 
Here is my command and output.(FMUV5 + Radio)`./set_actuator serial:///dev/ttyUSB0 1 0.6`
```
[09:51:23|Info ] MAVSDK version: v0.40.0 (mavsdk_impl.cpp:21)
Waiting to discover system...
[09:51:23|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:538)
[09:51:23|Debug] Component Autopilot (1) added. (system_impl.cpp:354)
[09:51:24|Debug] Setting UUID to: 3474897730398270777 (system_impl.cpp:222)
[09:51:24|Debug] Discovered 1 component(s) (UUID: 3474897730398270777) (system_impl.cpp:511)
Discovered autopilot
Setting actuator...
[09:51:24|Debug] MAVLink: info: Data link regained (system_impl.cpp:253)
```
I think it means MAIN1 (or MAIN 2) will output the PWM(value: 0.6), while there was nothing happened.
So, how can I make the actuator to rotate?
Thanks again!

Another small question: After updating PX4 to newtest(v1.10->v1.11 beta), the gazebo simulation encounterd some problem.
```
pxh> commander takeoff
pxh> WARN  [PreFlightCheck] Preflight Fail: Compasses -2147483648° inconsistent
WARN  [PreFlightCheck] Please check orientations and recalibrate
WARN  [PreFlightCheck] Preflight Fail: Baro Sensor 0 missing
INFO  [tone_alarm] notify negative
```

I did as here
#https://github.com/PX4/PX4-Autopilot/issues/12469
It's useless.
I changed it  to V1.10, it worked well. It puzzled me.

julianoes, 2021-05-24:
> Another small question

You probably need to `make clean`.

YANG-DEMIN, 2021-05-24:
Yes, I did it. But it is uesless.
By the way, I clone the latest code from PX4, it is the same problem. When i switched to v1.10, it worked....

YANG-DEMIN, 2021-05-24:
Now, what I want to know most is that which channle `set_actuator` mapped. Such as, `index=1` represent `AUX1 or MAIN1`, `value = 0 `represent `pwm=1750`.

And why i have more than 7 MIAN output and 4 AUX output, there is only 6 vaule for index?(case 1-6).

Thanks again!
Yang

julianoes, 2021-05-25:
@coderkalyan do you know?

YANG-DEMIN, 2021-06-03:
> @YANG-DEMIN @julianoes
> 
> As per PX4 mixing docs: https://docs.px4.io/master/en/concept/mixing.html
> [image]
> 
> By default, index = 1 represents `AUX1`, and only AUX1-3 are supported due to how the mixer is configured.
> 
> `value = 0` represents `pwm=1500`, `1 = 2000`, `-1 = 1000`.

I GOT IT !
Thank you very much !!!
Best wishes for you

Yang

closed 2021-06-03 by YANG-DEMIN