mavlink/MAVSDK #1445: Provide Autopilot/Server functionality for MAVLink services

bazfp, 2021-05-26:
It would be exceptionally useful to have MAVSDK provide server functionality for MAVLink microservices. 

There is currently a gap missing in MAVLink, the raw data types are provided by the MAVLink headers but the implementation of the microservices (Command, Mission, Heartbeat, Parameters etc) is still up to the user to implement. This can easily lead to fragmentation of the standard with the services. It also makes MAVLink integration into existing systems difficult.

MAVSDK does a good job of tackling this from a GCS/Companion side, but API is not currently suitable for the autopilot side as it was built on the assumption we're communicating with a MAVLink autopilot that implements the services we require. MAVSDK connects as a client.

There is a real case for server implementation in MAVSDK along these lines and MAVSDK could be used as a basis for a MAVLink autopilot. While obviously not for embedded platforms, there are certainly use-cases for this on non-PX4/Ardupilot autopilot systems but also in testing frameworks. In addition MAVSDK could bridge existing autopilot systems with MAVLink.

I propose MAVSDK implement the following plugins:

- Mission server (`MISSION_REQUEST_INT` workflow from autopilot side)
- Command server (subscribe to commands, ack, etc)
- Parameter server
- Message server (`MAV_CMD_REQUEST_MESSAGE`? `MAV_CMD_SET_MESSAGE_INTERVAL` support?)

I've recently documented what MAVLink messages + parameters MAVSDK requires in order to function, I can list these as it would probably be a useful starting point for these services.

bazfp, 2021-05-26:
Here is a big list of messages used by MAVSDK in action, telemetry & mission plugins.

**Parameters required to set health OK in Telemetry:**

* *CAL_GYRO0_ID* // Set to true once gyro is stabilised. (true if doesn't exist)
* *CAL_ACC0_ID* // Set to true once accelerometer is stabilised (true if doesn't exist)
* *CAL_MAG0_ID* // Set to true once magnetometer is stabilised (true if doesn't exist)

```
Telemetry Messages
------------------

+----------------------+----------------------+----------------------+----------------------+----------------------+
| Message              | ID                   | Direction            | Description          | Notes                |
+----------------------+----------------------+----------------------+----------------------+----------------------+
| HEARTBEAT            | #0                   | Bidirectional        | Heartbeat message is | Autopilot should     |
|                      |                      |                      | used to register     | emit heartbeats      |
|                      |                      |                      | MAVLink compatible   | periodically (>1hz). |
|                      |                      |                      | systems              | This is used to      |
|                      |                      |                      |                      | identify autopilot   |
|                      |                      |                      |                      | version and vehicle  |
|                      |                      |                      |                      | frame type (e.g. *M  |
|                      |                      |                      |                      | AV_TYPE_SURFACE_BOA  |
|                      |                      |                      |                      | T*)                  |
+----------------------+----------------------+----------------------+----------------------+----------------------+
| AUTOPILOT_VERSION    | #148                 | Autopilot to         | Version and          | Used to register     |
|                      |                      |                      | capability of        | autopilot system.    |
|                      |                      |                      | autopilot software.  | Requested via *MAV_  |
|                      |                      |                      |                      | CMD_REQUEST_AUTOPIL  |
|                      |                      |                      |                      | OT_CAPABILITIES*     |
+----------------------+----------------------+----------------------+----------------------+----------------------+
| HOME_POSITION        | #242                 | Autopilot to MAVSDK  | Send the vehicle     | Required for Health  |
|                      |                      |                      | home (recovery)      | OK                   |
|                      |                      |                      | position             |                      |
+----------------------+----------------------+----------------------+----------------------+----------------------+
| LOCAL_POSITION_NED   | #32                  | Autopilot to MAVSDK  | Vehicle position     | Must be sent at      |
|                      |                      |                      | relative to startup  | minimum 1hz if       |
|                      |                      |                      | global position      | available. Required  |
|                      |                      |                      |                      | for MAVSDK Health OK |
+----------------------+----------------------+----------------------+----------------------+----------------------+
| GPS_RAW_INT          | #24                  | Autopilot to MAVSDK  | Vehicle raw global   | Required for MAVSDK  |
|                      |                      |                      | position from GPS    | Health OK (>1 Hz)    |
|                      |                      |                      | sensor               | once GPS OK          |
+----------------------+----------------------+----------------------+----------------------+----------------------+
| GLOBAL_POSITION_INT  | #33                  | Autopilot to MAVSDK  | Vehicle              | Required for MAVSDK  |
|                      |                      |                      | filtered/estimated   | Health OK (>1 Hz)    |
|                      |                      |                      | global position      |                      |
+----------------------+----------------------+----------------------+----------------------+----------------------+
| GPS_GLOBAL_ORIGIN    | #49                  | Autopilot to MAVSDK  | GPS of local origin  | Required for MAVSDK  |
|                      |                      |                      | (local 0,0,0)        | Health OK            |
|                      |                      |                      | position             |                      |
+----------------------+----------------------+----------------------+----------------------+----------------------+

Parameter Messages
------------------

+----------------------+----------------------+----------------------+----------------------+----------------------+
| Message              | ID                   | Direction            | Description          | Notes                |
[... cut ...]
|                           |                           | this area.                |                           |
+---------------------------+---------------------------+---------------------------+---------------------------+
| MAV_CMD_DO_REPOSITION     | #192                      | Reposition the vehicle to | Is used for debugging     |
|                           |                           | target co-ordinates       | guidance.                 |
+---------------------------+---------------------------+---------------------------+---------------------------+

```

JonasVautherin, 2021-05-27:
@julianoes: About CAL_GYRO0_ID, CAL_ACC0_ID and CAL_MAG0_ID: wasn't that supposed to be replaced by the so-called "events interface" at some point? Do you know anything about that?

julianoes, 2021-05-27:
@JonasVautherin I think the calibration check and the telemetry health flags should be covered by [SYS_STATUS](https://mavlink.io/en/messages/common.html#SYS_STATUS).

bazfp, 2021-08-03:
Server functionality has been added: (after much hardwork :tada:)

`mission_raw_server` #1488
`action_server` https://github.com/mavlink/MAVSDK/pull/1491
`telemetry_server` https://github.com/mavlink/MAVSDK/pull/1482
`plugin_server` https://github.com/mavlink/MAVSDK/pull/1473

Documentation and examples required:

closed 2026-09-24 by julianoes