mavlink/MAVSDK #1458: Best way to work with qtcreator?

reedev, 2021-06-07:
Hi,
I would like to build all with qtcreator on Ubuntu (desktop) and I was wondering if someone can recommend a preferred way. At the moment I only see cmd line build instructions. The cmd line instructions are fine for building end user applications but now I would like to dive into the MAVSDK itself.
thanks!

julianoes, 2021-06-07:
The command line build instructions are just plain cmake and make commands. So if QtCreator supports cmake/make projects it should "just work".

Have you tried this? https://doc-snapshots.qt.io/qtcreator-4.0/creator-project-cmake.html

reedev, 2021-06-07:
Sure, I have tried it and I don't see an install target or so. This question was merely to ask for handy do's and don't's. Maybe others have suggestions and in the mean time I will make it work.

julianoes, 2021-06-07:
> I don't see an install target or so

What do you mean? Where do you want to install it?

reedev, 2021-06-07:
I don't see the target in qtcreator. Working from the cmd line works out-of-the-box (including the install prefix) so it's just a matter of finding a best way with qtcreator.

julianoes, 2021-06-07:
I tried this and did: Open Project -> Select CMakeLists.txt

Then it does the configure and build, and in the Projects tab you can add more configure options such as `BUILD_MAVSDK_SERVER` if required:
[image]

reedev, 2021-06-07:
yes, such things indeed. I am just not yet familiar with MAVSDK since the cmd line build 'just works' right away.

julianoes, 2021-06-29:
I'm going to close this. Please comment again if there are more questions.

closed 2021-06-29 by julianoes

Leduvanh44, 2024-04-07:
Hi,
I would like to build a qt app and use mavsdk to connect my AGV. How can i include mavsdk into my qt5 project ?
Thanks!

julianoes, 2024-04-07:
If you're using cmake, then use the way mavsdk is used in any of the [examples](https://github.com/mavlink/MAVSDK/tree/main/examples).