mavlink/MAVSDK #1463: drone.telemetry.scaled_pressure() hangs

olafwessler, 2021-06-07:
Hi there,

I am trying to get the atmospheric pressure from the drone but the code hangs forever at the async call without any error.

    # get atmospheric pressure at takeoff location
    async for pre in drone.telemetry.scaled_pressure():
        pressure=pre.absolute_pressure_hpa
        print("Pressure at takeoff location: %.2fmbar" %(pressure))
        break

Using mavsdk Python on Windows 
PX4 mini and mavsdk have been updated to latest version

thanks for help

JonasVautherin, 2021-06-07:
If you never receive any `scaled_pressure` in the `async for`, it suggests that PX4 does not send them. Could it be?

olafwessler, 2021-06-07:
That makes sense! Do you know about any plans to make those available to telemetry? Havent found any parameter to turn them on...

TSC21, 2021-06-07:
There's no parameter to enable it. They are supported but not sent by default. You will have to request the autopilot to stream the messages using MAVLink [MAV_CMD_SET_MESSAGE_INTERVAL](https://mavlink.io/en/messages/common.html#MAV_CMD_SET_MESSAGE_INTERVAL) or eanbling them by default using the `mavlink` module commands: https://dev.px4.io/master/en/middleware/modules_communication.html#mavlink.

olafwessler, 2021-06-07:
found it, thank you. But the drone's barometric pressure is about 4m different than my ground station, so this is not very useful...

closed 2021-06-07 by olafwessler

julianoes, 2021-06-10:
@olafwessler you might be able to correct/calibrate the baro altitude using a param. It might be this one: https://docs.px4.io/master/en/advanced_config/parameter_reference.html#SENS_BARO_QNH

olafwessler, 2021-06-10:
@julianoes I am using the absolute pressure value from both drone and ground station. So applying the QNH calibration to both devices will not eliminate the difference..
By the way: As it is possible to receive pressure data from mavsdk and to set a calibration value by parameter, would it make sense to also allow to activate the data stream itself from PX4 by parameter?

julianoes, 2021-06-11:
I think you would only apply the calibration to one but maybe I don't understand the problem correctly. I just assume that these pressure sensors are not perfectly calibrated which is probably what you're finding.

> activate the data stream itself from PX4 by parameter?

You mean so that you don't have to use `set_rate_scaled_pressure()`?

olafwessler, 2021-06-11:
@julianoes The QNH calibration is to adjust the pressure-to-altitude conversion according to local weather conditions. It does not affect the absolute pressure value. My goal was not to get the altitude based on pressure but to get a relative height difference between ground station and drone. So I am using absolute pressure values on both ends. 
But indeed acording to specs pressure sensors can have a tolerance of +/-10m.

Yes it would be good to be able to activate set_rate_scaled_pressure() by a parameter. The pressure value from mavsdk.telemetry cannot be used without the PX4 streaming this value. And to those only using the mavsdk without detailled knowledge of mavlink, a parameter would be more conveniant..

julianoes, 2021-06-11:
I was hoping the `set_rate` commands are convenient enough. That's why there here. The problem is that otherwise we need to add a parameter for each and every MAVLink message because everyone is going to want a specific one. :smile:

olafwessler, 2021-06-11:
I thought something like the SENS_EN_MB12XX parameter to activate a distance sensor would be very convenient. In particular, as the barometric pressure sensor is a standard build-in sensor in every pixhawk and the altitude value is already supported by mavsdk.
But I totally understand its getting too much at some point...