mavlink/MAVSDK #1470: how to access when flying mission

JonSnow-vn, 2021-06-19:
hi, i have a web control drone in jmavsim by click button, and button will run mavsdk module like takeoff, fly_mission, ...
when drone flying mission on jmavsim, how can i access to stop this by click button on web, t try to call land() in new module but get " bind error: Address already in use (udp_connection.cpp:86) "
How can i do this, thanks!

JonasVautherin, 2021-06-19:
So your button ends up starting  a new instance of MAVSDK? If you start two instances of MAVSDK, and both try to listen to udp:14540, then the second one will give you this `bind error: Address already in use (udp_connection.cpp:86)` error.

You need to have only one instance of MAVSDK, and all your buttons should talk to it :+1:

JonSnow-vn, 2021-06-19:
> So your button ends up starting a new instance of MAVSDK? If you start two instances of MAVSDK, and both try to listen to udp:14540, then the second one will give you this `bind error: Address [...]

ok i'm understend it, so how can i have only one instance of MAVSDK, or in other words when i click stop button stop how access instance of MAVSDK when flying, sorry i'm new ber

JonasVautherin, 2021-06-19:
I don't know what technology you use for your web interface or anything about your setup, I cannot guess :slightly_smiling_face:. With the information you shared, all I can say is that you cannot bind two instances of mavsdk to the same port.

JonSnow-vn, 2021-06-19:
> I don't know what technology you use for your web interface or anything about your setup, I cannot guess 🙂. With the information you shared, all I can say is that you cannot bind two instances of mavsdk to the same port.

Ok thanks. I have some solutions to slove this. It's work. But when the drone flying in missions, i'm called action.land() and drone was landed at current location. It's good. But some connect still on port 14540. So i cant do anything next like takeoff, flymission,.... It's say bind error: Address already in use (udp_connection.cpp:86) . How can i slove this?

julianoes, 2021-06-29:
> bind error: Address already in use (udp_connection.cpp:86) 

This means the UDP port is already used. So it seems like you are trying to connect with multiple processes to the drone. Is that true?

julianoes, 2024-05-27:
No answer, closing.

closed 2024-05-27 by julianoes