mavlink/MAVSDK #1472: Looks like RC loss, but RC is connected

imindude, 2021-07-01:
I made a tethered drone that was followed the another vehicle.
A few days ago, I found something strange thing.

Here is my Ulog.
https://logs.px4.io/plot_app?log=0e613e57-53b3-4ebe-bc3d-833980ce48be

And attached screenshot is telemetry log.
[image]

In Ulog, there is not shown the RC input - empty, but RC_CHANNELS in telemetry log shows RC is ok.
Actually, this drone is auto flight throughout the entire process. So, it does not need the RC. Just need the emergency status.

I have never seen it in past cases. Just one case.

My environment is,

PX4 v1.11.3
MAVSDK v0.40.0 (windows 10)

julianoes, 2021-07-01:
It might not show up as RC available because you might not have done the RC calibration?

In any case, I don't think this is a MAVSDK issue, right?

imindude, 2021-07-01:
> It might not show up as RC available because you might not have done the RC calibration?
> 
> In any case, I don't think this is a MAVSDK issue, right?

In next time, RC is completely worked. I did nothing between failure and success.
Actually I don't know it becomes from MAVSDK or PX4.

julianoes, 2021-07-01:
MAVSDK has nothing to do with RC_CHANNELS, pretty sure.

imindude, 2021-07-01:
> MAVSDK has nothing to do with RC_CHANNELS, pretty sure.

Yes. I think it's my mistake.

closed 2021-07-01 by imindude