mavlink/MAVSDK #1474: Building MAVSDK for Jetson Nano aarch64

dbaldwin, 2021-07-07:
Hi,

I'm hoping someone can help with a problem I've run into. The end goal is to get mavsdk_server running on my Jetson Nano. Then I'll use MAVSDK JavaScript to interface with a simulator (and then ultimately real hardware) from a web page. I've been following the instructions here:

https://mavsdk.mavlink.io/v0.24.0/en/contributing/build.html

to build for my Jetson Nano. I've gotten close with the following commands:

`docker run --rm dockcross/manylinux2014-aarch64 > ./manylinux2014-aarch`

`./dockcross-manylinux2014-aarch64 cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_MAVSDK_SERVER=ON -DBUILD_SHARED_LIBS=OFF -DBUILD_TESTS=OFF -Bbuild/manylinux2014-aarch64 -S.`

From here: https://hub.docker.com/r/dockcross/manylinux2014-aarch64

But then I ultimately end up with the following error:

[ 83%] Linking CXX static library libabsl_synchronization.a
[ 83%] Built target synchronization
gmake[3]: *** [all] Error 2
gmake[2]: *** [absl/src/absl-stamp/absl-build] Error 2
gmake[1]: *** [CMakeFiles/absl.dir/all] Error 2
gmake: *** [all] Error 2
**CMake Error at third_party/cmake/build_target.cmake:50 (message):
  /work/build/manylinux2014-aarch64/third_party/absl failed to build!**
Call Stack (most recent call first):
  third_party/CMakeLists.txt:23 (build_target)

-- Configuring incomplete, errors occurred!
See also "/work/build/manylinux2014-aarch64/CMakeFiles/CMakeOutput.log".

I've done some digging around in the GitHub issues and see a couple of references to absl, but I'm not 100% certain where to go from here. My compilation is being done on my MacBook using the main branch.

I'd also like to point out that building the linux-arm7 variant as mentioned in the documentation does work. 

Any insight would be very much appreciated.

Thanks.

JonasVautherin, 2021-07-08:
Doesn't [mavsdk_server_linux-arm64-musl](https://github.com/mavlink/MAVSDK/releases/download/v0.40.0/mavsdk_server_linux-arm64-musl) work on the Jetson Nano?

dbaldwin, 2021-07-08:
@JonasVautherin 🤦🤦🤦🤦🤦🤦🤦🤦🤦🤦

Sorry to waste your time with this. I don't know why I ended up in this rabbit hole. Actually I do remember now. I ran "uname -a" and it returned "aarch64". Anyway, the arm64-musl release did the track.

Thank you so much :)

closed 2021-07-08 by dbaldwin