mavlink/MAVSDK #1478: Cannot build examples on windows

fookinhell, 2021-07-09:


JonasVautherin, 2021-07-09:
Almost there, but I think you need to point CMake to the dependencies (curl is one of them), so something like this:

```
cmake -Bbuild -DCMAKE_PREFIX_PATH="../../install;../../build/default/third_party/install" -H.
```

fookinhell, 2021-07-09:
Well, that wasn't so hard, thanks :)

After removing the generated build folder and retrying to build the example, I know have this error : 
```
CMake Error at CMakeLists.txt:16 (add_executable):
  Target "takeoff_and_land" links to target "Threads::Threads" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
```

Shouldn't Threads be already linked ?

JonasVautherin, 2021-07-09:
Oh it's right that the example does [#include <thread>](https://github.com/mavlink/MAVSDK/blob/main/examples/takeoff_land/takeoff_and_land.cpp#L14), but does not link it (see [here](https://github.com/mavlink/MAVSDK/blob/main/examples/takeoff_land/CMakeLists.txt)).

The [fly_mission](https://github.com/mavlink/MAVSDK/blob/main/examples/fly_mission/CMakeLists.txt) example does it correctly. Would you mind changing takeoff_land to link threads, similarly?

fookinhell, 2021-07-09:
That did the trick !
Sorry for not taking the time to look at other examples.. I'll do it next time I have a problem :)

I'm closing this, thanks for your quick answers :)

closed 2021-07-09 by fookinhell

JonasVautherin, 2021-07-13:
> Sorry for not taking the time to look at other examples..

You actually found a bug in this one :slightly_smiling_face:. Would you mind opening a PR to fix it?