mavlink/MAVSDK #1504: Multiple telementry not accesible

saadnec, 2021-08-09:
I am trying to use multiple drones so i am assigning telementry1,2,3 for different drones to get positions of different drones.

****
     auto telemetry1 = Telemetry{system[0]};telemetry1::Position position1;
    auto action1 = Action{system[0]};
    auto offboard1 = Offboard{system[0]};
    auto telemetry2 = Telemetry{system[1]};telemetry2::Position position2;
    auto action2 = Action{system[1]};
    auto offboard2 = Offboard{system[1]};
    auto telemetry3 = Telemetry{system[2]};telemetry3::Position position3;
    auto action3 = Action{system[2]};
    auto offboard3 = Offboard{system[2]};
****

but i am getting an error
****
error: ‘telemetry1’ is not a class, namespace, or enumeration
  195 |     auto telemetry1 = Telemetry{system[1]};telemetry1::Position position1;
      |                                            ^~~~~~~~~~

error: ‘telemetry2’ is not a class, namespace, or enumeration
  197 |     auto telemetry2 = Telemetry{system[1]};telemetry2::Position position2;
      |                                            ^~~~~~~~~~
 error: ‘telemetry3’ is not a class, namespace, or enumeration
  200 |     auto telemetry3 = Telemetry{system[2]};telemetry3::Position position3;

***
if u could plz help me out to get position of the drones.

JonasVautherin, 2021-08-09:
The compiler complains because your code is not valid C++, it's independent from MAVSDK. I would suggest you have a look at the examples to see how we do it, e.g. [here](https://github.com/mavlink/MAVSDK/blob/main/examples/takeoff_and_land/takeoff_and_land.cpp#L92-L94).

saadnec, 2021-08-09:
> 
> 
> The compiler complains because your code is not valid C++, it's independent from MAVSDK. I would suggest you have a look at the examples to see how we do it, e.g. [here](https://github.com/mavlink/MAVSDK/blob/main/examples/takeoff_and_land/takeoff_and_land.cpp#L92-L94).

I have checked it out but exactly what should i do to get position of my different drones because in the examples we have either used a single drone or even if we have used multiple ones we have  done it using 1 function with 1 drone at a time, but i want to use multiple ones in a single function so i need telemetry of 3 different vehicles.

JonasVautherin, 2021-08-09:
Again that's a C++ question, not a MAVSDK one :blush:. You have multiple callbacks and you want to use them together. It's up to you to decide how you design that :+1:

saadnec, 2021-08-09:
So if u could please guide me about how we are receiving callbacks. 

****
telemetry.subscribe_position([](Telemetry::Position position) {
        std::cout << "Altitude: " << position.relative_altitude_m << " m\n";
    });
****
Here why are we passing Telemetry::Position not telemetry::Position  because we have initiated the system plugin by ""auto telemetry = Telemetry{system};""

bazfp, 2021-08-09:
> So if u could please guide me about how we are receiving callbacks.
>
> telemetry.subscribe_position([](Telemetry::Position position) {
> std::cout << "Altitude: " << position.relative_altitude_m << " [...]

This is a C++ question - these bug reports are for MAVSDK library issues.

`Telemetry::Position` is a data type. The instances you have created (`telemetry1` `telemetry2`) affect only how you setup the subscription.

```
// Setup for telemetry 1
telemetry1.subscribe_position([](Telemetry::Position position) {
std::cout << "Altitude 1: " << position.relative_altitude_m << " m\n";
});`

// Setup for telemetry 2
`telemetry2.subscribe_position([](Telemetry::Position position) {
std::cout << "Altitude 2: " << position.relative_altitude_m << " m\n";
});
```

The only difference is the variable you call `subscribe_position` onto :)

https://stackoverflow.com/questions/7627098/what-is-a-lambda-expression-in-c11

closed 2021-08-09 by bazfp

saadnec, 2021-08-11:
> 
> 
> > So if u could please guide me about how we are receiving callbacks.
> > telemetry.subscribe_position([](Telemetry::Position position) {
> > std::cout << "Altitude: " << position.relative_altitude_m << " m\n";
> > });
> > Here why are we passing Telemetry::Position not telemetry::Position because we have initiated the system plugin by ""auto telemetry = Telemetry{system};""
> 
> This is a C++ question - these bug reports are for MAVSDK library issues.
> 
> `Telemetry::Position` is a data type. The instances you have created (`telemetry1` `telemetry2`) affect only how you setup the subscription.
> 
> ```
> // Setup for telemetry 1
> telemetry1.subscribe_position([](Telemetry::Position position) {
> std::cout << "Altitude 1: " << position.relative_altitude_m << " \n";
> });`
> 
> // Setup for telemetry 2
> telemetry2.subscribe_position([](Telemetry::Position position) {
> std::cout << "Altitude 2: " << position.relative_altitude_m << " \n";
> });
> ```
> 
> The only difference is the variable you call `subscribe_position` onto :)
> 
> https://stackoverflow.com/questions/7627098/what-is-a-lambda-expression-in-c11

Thanks. This was helpful :). 

Can you please guide me about at what point is the subscribe_position is called as it continues to print till the program isn't terminated.