mavlink/MAVSDK #1505: Question about Pixhawk Cube Orange & Pixhawk Cube Black

marcinm97, 2021-08-11:
Does the MAVSDK C++ support both versions of pixhawk? ( Orange Cube & Black Cube )

reedev, 2021-08-12:
I guess it might be good to describe what you are trying to achieve. What is your setup wrt PX4 or ArduPilot, etc. What have you tried so far? Giving more details will allow people to help you better.

JonasVautherin, 2021-08-12:
MAVSDK supports mainly PX4, and I have seen people use at least parts of it with Ardupilot.

It's not related to pixhawk though: pixhawk is the hardware, and MAVSDK does not care about the hardware. So you can run PX4 on a pixhawk, a raspberry pi, your laptop or a coffee machine, as long as MAVLink messages reach MAVSDK, it will be fine.

Does that answer your question? :blush:

julianoes, 2021-08-30:
Closing as there was no follow up.

closed 2021-08-30 by julianoes