mavlink/MAVSDK #1506: Action not working properly

saadnec, 2021-08-11:
I just added action.gotolocation in the takeoff example but as soon as it takes this command instead of going to that location my vehicle starts  going down and lands.

Pictures of codes and mavsdk messages are attached 
[image]
[image]

I added these two lines
***
sleep_for(seconds(10));
    const Action::Result result = action.goto_location(47.3974,8.54565,2.5,0);
    if (result != Action::Result::Success) {
        std::cerr << "Action failed:" << takeoff_result << '\n';
    }
***

what am i doing wrong ?

reedev, 2021-08-12:
Never tried this particular case myself but I know from missions that you need to be within a certain distance (900 meters) from the home point. I guess you need to set the home position, see for sitl https://docs.px4.io/master/en/simulation/gazebo.html

JonasVautherin, 2021-08-12:
The altitude of the goto_location command is given above sea level (see the documentation here: https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/action/action.proto#L79-L80).

So if your drone takes off 400m above the sea level, and you ask it to go to 2.5 meters above sea level, then it will go down. Could that be the reason for your issue?

saadnec, 2021-08-13:
> The altitude of the goto_location command is given above sea level (see the documentation here: https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/action/action.proto#L79-L80).
>
> So if your [...]

Yes, that was the issue. Thanks for helping me resolve this issue :).

closed 2021-08-13 by saadnec