mavlink/MAVSDK #1507: Testing an 'Unsupported component' with sitl?

reedev, 2021-08-12:
First of all: I am learning about the discovery of components and I see that SystemImpl::add_new_component() is constantly called. Obviously, components are dynamically checked at runtime. Is that a right assumption?

Since I have an 'Unsupported component' from my setup Herelink/Pixhawk4/PX4 I would like to check if that is the reason why component discovery does not work as with sitl. Is there any way to do this in sitl?

JonasVautherin, 2021-08-12:
> Obviously, components are dynamically checked at runtime. Is that a right assumption?

Yes, that's correct.

> I would like to check if that is the reason why component discovery does not work as with sitl

What does not work? Like what kind of code do you run that works with SITL and not with your setup?

reedev, 2021-08-12:
Hi,

I don't understand how new components are added, and seemingly removed. Running any examples with SITL (Gazebo) works fine and none of them work with my basic setup. For example, when running the takeoff_and_land the autopilot is not found.

I added a few lines for debugging at line 357 in system_impl.cpp:
    LogDebug() << "  Num components before inserting: " << _components.size();
    auto res_pair = _components.insert(component_id);
    LogDebug() << "  Num components after inserting: " << _components.size();

And this is the output:

[07:08:00|Info ] MAVSDK version: v0.42.0-dirty (mavsdk_impl.cpp:29)
[07:08:00|Info ] New system on: 192.168.2.7:14552 (with sysid: 0) (udp_connection.cpp:191)
[07:08:00|Debug] New: System ID: 0 Comp ID: 0 (mavsdk_impl.cpp:468)
[07:08:00|Debug] Discovered 0 component(s) (system_impl.cpp:514)
Waiting to discover system...
[07:08:00|Debug]   Num components before inserting: 0 (system_impl.cpp:357)
[07:08:00|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
[07:08:00|Debug] Component Autopilot (1) added. (system_impl.cpp:367)
[07:08:00|Debug]   Num components before inserting: 1 (system_impl.cpp:357)
[07:08:00|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
[07:08:00|Debug]   Num components before inserting: 1 (system_impl.cpp:357)
[07:08:00|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
[07:08:00|Debug]   Num components before inserting: 1 (system_impl.cpp:357)
[07:08:00|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
[07:08:00|Debug]   Num components before inserting: 1 (system_impl.cpp:357)
[07:08:00|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
[07:08:00|Debug]   Num components before inserting: 1 (system_impl.cpp:357)
[07:08:00|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
[07:08:00|Debug]   Num components before inserting: 1 (system_impl.cpp:357)
[07:08:00|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
[07:08:00|Debug]   Num components before inserting: 1 (system_impl.cpp:357)
[07:08:00|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
[07:08:00|Debug]   Num components before inserting: 1 (system_impl.cpp:357)
[07:08:00|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
[07:08:00|Debug]   Num components before inserting: 1 (system_impl.cpp:357)
[07:08:00|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
[07:08:00|Debug]   Num components before inserting: 1 (system_impl.cpp:357)
[07:08:00|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
[07:08:00|Debug]   Num components before inserting: 1 (system_impl.cpp:357)
[07:08:00|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
[07:08:00|Debug]   Num components before inserting: 1 (system_impl.cpp:357)
[07:08:00|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
[07:08:01|Debug] New: System ID: 42 Comp ID: 250 (mavsdk_impl.cpp:468)
[07:08:01|Debug]   Num components before inserting: 0 (system_impl.cpp:357)
[07:08:01|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
[07:08:01|Debug] Component Unsupported component (250) added. (system_impl.cpp:367)
[07:08:01|Debug]   Num components before inserting: 1 (system_impl.cpp:357)
[07:08:01|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
[07:08:01|Debug]   Num components before inserting: 1 (system_impl.cpp:357)
[07:08:01|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
[07:08:01|Debug]   Num components before inserting: 1 (system_impl.cpp:357)
[07:08:01|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
[07:08:01|Debug]   Num components before inserting: 1 (system_impl.cpp:357)
[07:08:01|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
[07:08:01|Debug]   Num components before inserting: 1 (system_impl.cpp:357)
[07:08:01|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
[07:08:01|Debug]   Num components before inserting: 1 (system_impl.cpp:357)
[07:08:01|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
[07:08:01|Debug]   Num components before inserting: 1 (system_impl.cpp:357)
[07:08:01|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
[07:08:01|Debug]   Num components before inserting: 1 (system_impl.cpp:357)
[07:08:01|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
[07:08:01|Debug]   Num components before inserting: 1 (system_impl.cpp:357)
[07:08:01|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
[07:08:01|Debug]   Num components before inserting: 1 (system_impl.cpp:357)
[07:08:01|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
[07:08:01|Debug]   Num components before inserting: 1 (system_impl.cpp:357)
[07:08:01|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
[07:08:01|Debug]   Num components before inserting: 1 (system_impl.cpp:357)
[... 108 lines cut ...]
[07:08:02|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
[07:08:02|Debug]   Num components before inserting: 2 (system_impl.cpp:357)
[07:08:02|Debug]   Num components after inserting: 2 (system_impl.cpp:359)
[07:08:02|Debug]   Num components before inserting: 1 (system_impl.cpp:357)
[07:08:02|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
[07:08:03|Warn ] sending again after 1.51535 s, retries to do: 1  (520). (mavlink_commands.cpp:231)
[... cut ...]
[07:08:03|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
[07:08:03|Debug]   Num components before inserting: 1 (system_impl.cpp:357)
[07:08:03|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
[07:08:03|Debug]   Num components before inserting: 1 (system_impl.cpp:357)
[07:08:03|Debug]   Num components after inserting: 1 (system_impl.cpp:359)
No autopilot found.
[07:08:03|Debug]   Num components before inserting: 2 (system_impl.cpp:357)
[07:08:03|Debug]   Num components after inserting: 2 (system_impl.cpp:359)

JonasVautherin, 2021-08-13:
MAVSDK says "Component Autopilot (1) added."

What says "No autopilot found."?

reedev, 2021-08-13:
All the info is from the MAVSDK internal and the takeoff_and_land.cpp Please check line 52.

reedev, 2021-08-18:
@JonasVautherin any idea how this might happen?

julianoes, 2021-08-30:
I don't quite understand this yet. So what exactly is the problem that you are trying to solve?

reedev, 2021-08-30:
No problem, I will give you more details. I am trying to connect to a Pixhawk 4 running PX4 via a Herelink controller. Since I want to understand the underlying code as much as possible, I thought I'd ask a sitl question instead of directly asking about my particular setup. As always, it would be better to replicate a hardware related issue in sitl and fix it there first.

In the long log of the connection with the devices you can see the line
[07:08:00|Debug] Component Autopilot (1) added. (system_impl.cpp:367)
but eventually the take_off_and_land example concludes that no autopilot is found...

I hope this clearifies it a bit more... thanks

julianoes, 2021-08-30:
Got it. Can you show the code that runs the output above?

reedev, 2021-08-30:
Ok cool. It is the standard example takeoff_and_land and I added a few lines for debugging at line 357 in system_impl.cpp:
LogDebug() << " Num components before inserting: " << _components.size();
auto res_pair = _components.insert(component_id);
LogDebug() << " Num components after inserting: " << _components.size();

This is actually the same info as I wrote above. Do you need more info?

I was wondering why components are 'lost' during discovery, i.e. num components goes from 2 to 1...

julianoes, 2021-08-31:
> It is the standard example takeoff_and_land

That's what I needed to know, thanks.

They should not get "lost", that's indeed odd.

reedev, 2021-08-31:
Perfect. Plz let me know how I can help.

julianoes, 2021-08-31:
Aha!

So this another `System` (and `SystemImpl`):
```
[07:08:01|Debug] New: System ID: 42 Comp ID: 250 (mavsdk_impl.cpp:468)
[07:08:01|Debug] Num components before inserting: 0 (system_impl.cpp:357)
```

So this explains why the component number "goes down".

Now, there is still the question of why there is no autopilot found. Given it says:
```
[07:08:00|Debug] Component Autopilot (1) added. (system_impl.cpp:367)
```

it really should be there, so for some reason this doesn't get triggered.

In theory `mavsdk.systems().back()` should always give you the one that was just added, and so one of the two should fulfill the promise in:

https://github.com/mavlink/MAVSDK/blob/61eb1f14151b95385869a6691353ec236e693a14/examples/takeoff_and_land/takeoff_and_land.cpp#L35-L54

reedev, 2021-09-14:
@julianoes  Just to let you know I am still on it (every now and then) and I see what you mean, about the different systems. Since I am still fresh I did not realize this... Thanks for explaining.

The lambda is not called and I am looking into this, learning about the MAVSDK internals on the fly... (I changed the mav sys id to 42 too, to keep it simpler, hopefully)

reedev, 2021-09-14:
How can I use a sitl running on a remote system so that I need to connect via a connection string like udp://192.168.1.2:14550 or similar? I am asking because the above issue occurs when I make a connection like that and it does not occur when I use the server config like udp://:14550 (that's right, server, isn't it?).

MavsdkImpl::setup_udp_remote() is called of course and that path doesn't seem to work...

reedev, 2021-09-14:
Update: although good to know, it is not essential for this issue since I experience it too when a remote QGC (on the Herelink controller) connects to our MAVSDK server...

julianoes, 2021-09-15:
> How can I use a sitl running on a remote system so that I need to connect via a connection string like udp://192.168.1.2:14550 or similar?

On that SITL system you need to send UDP to the right IP, see:
https://docs.px4.io/master/en/simulation/#enable-streaming-to-specific-address

julianoes, 2021-09-20:
@reedev this might help your debugging: https://github.com/mavlink/MAVSDK/pull/1557

reedev, 2021-09-22:
As it turned out, the System did not reply quickly enough, or did not send a heartbeat in time (did not perform any specific tests) and therefore the System was not checked in the callback. Changing the get_system() into:
std::shared_ptr<System> get_system(Mavsdk& mavsdk)
{
    std::cout << "Waiting to discover system...\n";
    auto prom = std::promise<std::shared_ptr<System>>{};
    auto fut = prom.get_future();

    // We wait for new systems to be discovered, once we find one that has an
    // autopilot, we decide to use it.
    mavsdk.subscribe_on_new_system([&mavsdk, &prom]() {
        for (auto& system : mavsdk.systems()) {
            if (system->has_autopilot()) {
                std::cout << "Discovered autopilot\n";

                // Unsubscribe again as we only want to find one system.
                mavsdk.subscribe_on_new_system(nullptr);
                prom.set_value(system);
            }
        }
    });

    // We usually receive heartbeats at 1Hz, therefore we should find a
    // system after around 3 seconds max, surely.
    if (fut.wait_for(seconds(3)) == std::future_status::timeout) {
        std::cerr << "No autopilot found.\n";
        return {};
    }

    // Get discovered system now.
    return fut.get();
}
makes sure that the correct system is found anyway.

Thanks for your help!

closed 2021-09-22 by reedev

JonasVautherin, 2021-09-22:
@reedev I am not sure I understand your fix. Are you saying that something is wrong in the example code? Would you mind opening a PR to fix it, if that's the case? Would be helpful :innocent:

reedev, 2021-09-22:
@JonasVautherin no something is wrong, or at least different in my setup. I tested again with the latest revision (with changed connection code compared to the version that I used for my first tests) and then I saw that only the 2nd system is being processed in the callback, and not the first one. So my code is only a fix for my setup, I guess, and I wanted to drop this knowledge here for future reference for anyone else.

I did not get to the bottom of the issue in - most likely - my system and I will use this pragmatic approach for the moment.

julianoes, 2021-09-23:
@JonasVautherin I think the issue is connected to the fact that `is_connected` and `has_autopilot` are not necessarily in sync.

So, when any message from sysid 1, compid 1 arrives, you will be told that `has_autopilot` is true, however, it is not connected yet, until a heartbeat arrives.

To fix this we could consider not accepting any message until a heartbeat from a sysid/compid is received.

reedev, 2021-09-23:
I have seen recent code to log mavlink messages. Is that right? Could that help somehow to backup this assumption?

julianoes, 2021-09-23:
You can use `export MAVSDK_MESSAGE_DEBUGGING=1` to get more information about messages arriving and being sent.