mavlink/MAVSDK #1508: (slightly off-topic) How to 'reset' Gazebo after running integrations tests?

reedev, 2021-08-13:
Hi,

v0.42.0
I ran the integrations tests and now running the  MAVSDK example fly_mission (with custom waypoints next to the home point) and now I get "MAVLink: critical: Geofence violation for waypoint". Which is understandable since I ran the integration tests...

Is there a way to reset/clear Gazebo? I couldn't find an answer in the MAVSDK issues...

Thanks

JonasVautherin, 2021-08-13:
I just restart it :sweat_smile:. I don't think there is a way, to be honest

reedev, 2021-08-13:
No way, that doesn’t work for me...?! Of course, I tried that 😛

mrpollo, 2021-08-14:
I think what @JonasVautherin means is you need to stop PX4 and relaunch it. That should start the gazebo simulation environment in a fresh state.

reedev, 2021-08-14:
Hi Ramon,
Yes, that’s how I interpreted Jonas’ suggestion, thanks. I also restarted the machine and the geofence violation was still there.

I now realized that it is indeed px4 that does the sitl, thanks.

I remember that I also ran px4 unit tests and integrations tests so maybe the state is stored in the px4 source tree. Not sure, I am learning about px4, mavsdk, etc

I have now ‘fixed’ the issue by using a fresh clone and build of the px4…

mrpollo, 2021-08-14:
It's possible that PX4 stores some values to make the build faster, in those cases I recommend you run 'make clean' it's faster than a fresh clone

reedev, 2021-08-16:
Fyi, I tried that but no change. Actually, I did not expect this to be honest since 'make clean' is a build command. Anyway, it's not important for now.

julianoes, 2021-08-30:
O

reedev, 2021-08-30:
Yes, that sounds right to me too.
For my understanding, which file stores the geofence, do you know that? Is there documentation about these sitl runtime files?

julianoes, 2021-08-30:
This should be the file: `build/px4_sitl_default/tmp/rootfs/dataman`.

reedev, 2021-08-30:
All clear, thanks

PR #1517 (2021-08-31): geofence: add method to clear fence
This adds the functionality to clear geofence existing on the vehicle.

Fixes #1508.
closed 2021-08-31 by julianoes