mavlink/MAVSDK #1522: Waypoint frame not supported

FurkanYelbey, 2021-09-01:
Waypoint frame not supported (mission_impl.cpp:529)
I got that error everytime I try to execute cpp in my ubuntu it's happening when I try download_mission command how can I fix that and what does that mean guys? I can't find any solutions for 2 months

JonasVautherin, 2021-09-02:
Could you share code that we could use to reproduce your issue?

> I can't find any solutions for 2 months

What have you tried already? If you spent 2 months on it you may have valuable information to share with us and help debug the issue, so feel free to share! :blush:

julianoes, 2021-09-02:
What Jonas said. @FurkanYelbey are you using PX4 or ArduPilot? What mission are you uploading and how?

FurkanYelbey, 2021-09-02:
I am using PX4 and trying to download some waypoints on my PX4 but it gives me Waypoint frame not supported

#include <mavsdk/mavsdk.h>
#include <mavsdk/plugins/action/action.h>
#include <mavsdk/plugins/mission/mission.h>
#include <mavsdk/plugins/telemetry/telemetry.h>

#include <chrono>
#include <functional>
#include <future>
#include <iostream>
#include <memory>
#include <thread>

using namespace mavsdk;
using namespace std;
using std::chrono::seconds;
using std::this_thread::sleep_for;

void usage(const std::string& bin_name)
{
	cerr << "Usage : " << bin_name << " <connection_url>\n";
	// biz buraya Serial ba?lant? girece?iz o da ?u ?ekilde
	// serial:///path/to/serial/dev[:baudrate]
}

shared_ptr<System> get_system(Mavsdk& mavsdk)
{
	cout << "Waiting to discover system...\n";
	auto prom = promise<shared_ptr<System>>{};
	auto fut = prom.get_future();

	mavsdk.subscribe_on_new_system([&mavsdk, &prom]() {
		auto system = mavsdk.systems().back();

		if (system->has_autopilot()) {
			cout << "Discovered autopilot\n" << endl;

			mavsdk.subscribe_on_new_system(nullptr);
			prom.set_value(system);
		}
		});

	//genelde heartbeat'ler 1Hz'de al?n?r biz de sistemi bulmas?n?
	//max 3 saniyede istiyoruz
	if (fut.wait_for(seconds(3)) == future_status::timeout) {
		cerr << "No autopilot found.\n";
		return{};
	}

	//sistemi ?imdi bul
	return fut.get();
}

int main(int argc, char** argv)//argc ve argv silindi
{

	Mavsdk mavsdk;

	string connection_url = "serial:///dev/ttyUSB0:57600";
	ConnectionResult connection_result = mavsdk.add_any_connection(connection_url);

	auto system = get_system(mavsdk);
	if (!system) {
		return 1;
	}

	auto action = Action{ system };
	auto mission = Mission{ system };
	auto telemetry = Telemetry{ system };

	while (!telemetry.health_all_ok()) {
		cout << "Waiting for system to be ready\n";
		sleep_for(seconds(1));
	}
	cout << "System ready\n";
	Mission::MissionPlan mission_plan{};

	cout << "Downloading mission from mission plan" << endl;

	std::cout << "Downloading mission." << std::endl;

	const pair<Mission::Result, Mission::MissionPlan> download_result = mission.download_mission();

	if (download_result.first != Mission::Result::Success) {
		cerr << "Mission download failed: " << download_result.first << ", exiting.\n";
		return 1;
	}

	if (download_result.second.mission_items.size() == 0) {
		cerr << "No missions! Exiting...\n" << endl;
		return 1;
	}

	cout << "Mission list: " << download_result.second << endl;

	const Mission::Result upload_result = mission.upload_mission(mission_plan);

	if (upload_result != Mission::Result::Success) {
		cerr << "Mission upload failed: " << upload_result << ", exiting. \n";
		return 1;
	}

	cout << "Arming... \n";
	const Action::Result arm_result = action.arm();
	if (arm_result != Action::Result::Success) {
		cerr << "Arming failed: " << arm_result << endl;
		return 1;
	}
	cout << "Armed.\n";

	auto prom = promise<void>{};
	auto fut = prom.get_future();

	//Görev ba?lamadan önce görev ilerleyi?ine subscribe oluyoruz
	mission.subscribe_mission_progress([&prom](Mission::MissionProgress mission_progress) {
		cout << "Mission progress update: " << mission_progress.current << " / "
			<< mission_progress.total << endl;
		if (mission_progress.current == mission_progress.total) {
			prom.set_value();
		}
		});

	Mission::Result start_mission_result = mission.start_mission();
	if (start_mission_result != Mission::Result::Success) {
		cerr << "Starting mission failed: " << start_mission_result << endl;
		return 1;
	}

	while (!mission.is_mission_finished().second) {
		sleep_for(seconds(1));
	}

	//Görev tamamland? RTL komutu ver
	cout << "Commanding RTL...\n";
	const Action::Result rtl_result = action.return_to_launch();
	if (rtl_result != Action::Result::Success) {
		cout << "Failed to command RTL: " << rtl_result << endl;
		return 1;
	}
	cout << "Commanded RTL.\n" << endl;

	//Biraz beklememiz gerekiyor yoksa armed durumu do?ru gerçekle?meyebilir
	sleep_for(seconds(2));

	while (telemetry.armed()) {
		//Tamamlanana kadar bekle
		sleep_for(seconds(1));
	}
	cout << "Disarmed, exiting.\n";

	cout << "Clearing mission from device" << endl;

	const Mission::Result clear_result = mission.clear_mission();

	if (clear_result != Mission::Result::Success) {
		cerr << "Clearing mission failed: " << clear_result << ", exiting.\n";
		return 1;
	}

}

that's my cpp file i tried to update my px4 tried to update MP i changed my pixhawk it was pixhawk 2.4.8 before and now it's Pixhawk4 but it still gives me that Unsupported error. Once time i don't know how did I do but it gave me time out but not not supported error

julianoes, 2021-09-02:
Which PX4 version? This would be something like v1.12. And which MAVSDK version?
And how are you creating and uploading the mission in the first place.

FurkanYelbey, 2021-09-02:
How can I learn my PX4 version and I use mavsdk 0.42.0 version

FurkanYelbey, 2021-09-02:
I am creating missions by Mission Planner and write them to PX4.
[image]

FurkanYelbey, 2021-09-02:
All I can see is firmware is 4.0.7 PX4

JonasVautherin, 2021-09-02:
MAVSDK is failing because it does not support a MAV_CMD_NAV_WAYPOINT that has a frame other than MAV_FRAME_GLOBAL_RELATIVE_ALT_INT. Could you tell us which frame MissionPlanner sent?

I don't know if PX4 supports other frames, btw, but I'm assuming that it does since you could upload it.

> All I can see is firmware is 4.0.7 PX4

I don't get that, the latest PX4 version I know is 1.12.x :sweat_smile:

julianoes, 2021-09-03:
Aha, you must be using ArduPilot, so that explains the issue. MAVSDK is not compatible with ArduPilot (yet) but there is work going on to fix that. Looks like we just found another thing that needs work. It has to do with the frame that you select in mission planner.

Actually, as an alternative you could just use `MissionRaw` instead of `Mission` which gives you more flexibility and direct access to the MAVLink  mission item type.

JonasVautherin, 2021-09-03:
> Actually, as an alternative you could just use MissionRaw instead of Mission which gives you more flexibility and direct access to the MAVLink mission item type.

This. And also you could try to change the frame in mission planner (let us know if you manage to do that).

This said, adding support for the Ardupilot frame in MAVSDK may be an relatively easy contribution, I believe. Your first step would be to understand which frame Ardupilot supports, and how it differs from MAV_FRAME_GLOBAL_RELATIVE_ALT_INT (as in: how could you convert from the Ardupilot one to MAV_FRAME_GLOBAL_RELATIVE_ALT_INT). Then we could add support for it in MAVDSK.

Does that make sense?

FurkanYelbey, 2021-09-03:
I'm gonna look for it how i can change the frame in mission planner and gonna try with MissionRaw