mavlink/MAVSDK #1525: How to get PositionNed from MAVSDK?

happyzihao, 2021-09-03:
Hello, I have a question and need help. 

from Mavsdk, I can get postion( latitude_deg, longitude_deg .. ) by function postion() in telemetry plugin.
Now, I want to get PostionNed( north_m, east_m .. ) , But telemetry plugin does not provide function positionned().

How can I get PostionNed ?

thanks.

julianoes, 2021-09-03:
This should have it:

https://github.com/mavlink/MAVSDK/blob/6ded0e9d7a2967ae9a59dfc05d1f499c7143f5b3/src/plugins/telemetry/include/plugins/telemetry/telemetry.h#L1386-L1389

happyzihao, 2021-09-04:
> This should have it:
> 
> https://github.com/mavlink/MAVSDK/blob/6ded0e9d7a2967ae9a59dfc05d1f499c7143f5b3/src/plugins/telemetry/include/plugins/telemetry/telemetry.h#L1386-L1389

thanks, it's ok.

closed 2021-09-04 by JonasVautherin