mavlink/MAVSDK #1527: Promise already satisfied

marvin-wtt, 2021-09-03:
I am running MAVSDK v.0.42.0 on Ubuntu 20.04 inside a ROS2 node. Sometimes the node dies right after I initialize MAVSDK. The error however occurs non deterministic with a 1/20 chance. (I just started it over and over again and sometimes it crashes)
After a lot of try and error debugging, I noticed that it happens every time `set_rate_battery` fails. I couldn't figure out why it fails but every time it does I get the error `Promise already satisfied`.

```
[my-node-1] [INFO] [1630672509.497487560] [uuid63085.fcc_brg_node]: Setting rate of battery failed
[my-node-1] terminate called after throwing an instance of 'std::future_error'
[my-node-1]   what():  std::future_error: Promise already satisfied
[ERROR] [my-node-1]: process has died [pid 145033, exit code -6, cmd '/home/user/my-project/install/my_node/lib/my_node/my_node --ros-args -r __node:=my_node_node -r __ns:=/uuid63085 --params-file /tmp/launch_params_vzif_yms'].
```

```C++
mavsdk_check_result_telemetry(telemetry_->set_rate_battery(2.0), "battery");
telemetry_->subscribe_battery([this](mavsdk::Telemetry::Battery battery) {
    auto msg = std_msgs::msg::Float32();
    msg.data = battery.remaining_percent;
    pub_state_battery_->publish(msg);
});
```
```C++
void MyClass::mavsdk_check_result_telemetry(mavsdk::Telemetry::Result result, string type)
{
    if (result != mavsdk::Telemetry::Result::Success) {
        RCLCPP_INFO(this->get_logger(), "%s", ("Setting rate of " + type + " failed").c_str());
    } else {
        RCLCPP_INFO(this->get_logger(), "%s", ("Setting rate of " + type + " succeeded").c_str());
    }
}
```

JonasVautherin, 2021-09-03:
Do you use futures/promises in your code? Or is this happening in the MAVSDK side?

marvin-wtt, 2021-09-03:
I do not use any futures in my code. I used a future in the system discovery before (like in the example) but I replaced it with busy waiting to narrow down the problem. 
So it must be on MAVSDK side as I could not imagine any side-effects with ROS2. 
It happens always the the result of `telemetry_->set_rate_battery()` is not success.

bazfp, 2021-09-03:
I have had this happen too when using the sync functions. It seems to be related to retrying commands

marvin-wtt, 2021-09-03:
I did some further testing and figured out, that the result is `CommandDenied`before the error occurs. I can't say if the behavior is unique to this result though. 
I am currently testing it in SITL but it also happens when connected to a flight controller. 
To reproduce the described behavior I have a disarmed copter in SITL and restart the software until I get the error. It takes approximately 20 restarts until the error occurs - sometimes even more.

julianoes, 2021-09-07:
I have seen this happen in CI tests recently as well. Could not track it down yet.

fas2037, 2021-09-08:
After some investigating I tracked the error down to: https://github.com/mavlink/MAVSDK/blob/f4987b17a0f766969efb2593e28d57c2c46c000e/src/core/mavlink_commands.cpp#L48-L56 It seems as the first result is a ```CommandDenied``` as mentioned above. However, just a moment later another result of the type ```Success``` is provided. This leads to ```std::future_error: Promise already satisfied```.  During further investigating I found this LogWarn Message in a comment in the ```do_work()``` function, which handles the ```_work_queue```: https://github.com/mavlink/MAVSDK/blob/f4987b17a0f766969efb2593e28d57c2c46c000e/src/core/mavlink_commands.cpp#L290-L303
After uncommentating the Warning Message I discovered, that the Error always occurs after this Warning was sent. Makes sense, as a the result is set to ```CommandDenied``` in the same if-statement.

After also uncommenting https://github.com/mavlink/MAVSDK/blob/f4987b17a0f766969efb2593e28d57c2c46c000e/src/core/mavlink_commands.cpp#L99-L100 and https://github.com/mavlink/MAVSDK/blob/f4987b17a0f766969efb2593e28d57c2c46c000e/src/core/mavlink_commands.cpp#L280 one can see these Logging messages:
```
[11:31:25|Debug] Command 511 to send to 1, 1 (mavlink_commands.cpp:99)
[11:31:25|Debug] sending it the first time (511) (mavlink_commands.cpp:280)
[11:31:25|Debug] sending it the first time (511) (mavlink_commands.cpp:280)
[11:31:25|Warn ] A command cannot be sent if another command with the same command_id is still processing! Ignoring command 511 (mavlink_commands.cpp:291)
[11:31:25|Debug] sending it the first time (511) (mavlink_commands.cpp:280)
terminate called after throwing an instance of 'std::future_error'
what():  std::future_error: Promise already satisfied
```

In my opinion this is due to the fact, that ```do_work()``` realizes, that another command with the same ID exists and ignores it but why does it return ```CommandDenied```? Shouldn't it just skip and and during the next iteration (or whenever the previous command with the same ID is done) it takes care of this command?

I tried it by commenting out https://github.com/mavlink/MAVSDK/blob/f4987b17a0f766969efb2593e28d57c2c46c000e/src/core/mavlink_commands.cpp#L295-L300 and setting the rate worked consistently (at least for 100 consecutive tries). During these 100 tries the Warning due to another command with the same ID still being processed occured 8 times, which also leads to the question why this even happens in the first place.

JonasVautherin, 2021-09-09:
It is expected that the second command receives a `CommandDenied` result. We don't want to retry it later, we delegate that to the user.

Now the thing is that there are two separate commands, and they should use two separate promises. Your bug suggests that maybe there is a confusion somewhere and that both results are sent to the same command :thinking:. Could it be?

julianoes, 2021-09-09:
> Now the thing is that there are two separate commands, and they should use two separate promises. Your bug suggests that maybe there is a confusion somewhere and that both results are sent to the same command thinking. Could it be?

That's an interesting thought.

julianoes, 2021-09-09:
@fas2037 do you have easy steps or a setup where it's easier to trigger this?

fas2037, 2021-09-09:
@julianoes with this setup I got the error for approximately 5% of my tries:
```cpp
#include <mavsdk/mavsdk.h>
#include <mavsdk/plugins/action/action.h>
#include <mavsdk/plugins/geofence/geofence.h> 
#include <mavsdk/plugins/mavlink_passthrough/mavlink_passthrough.h>
#include <mavsdk/plugins/mission_raw/mission_raw.h> 
#include <mavsdk/plugins/telemetry/telemetry.h>
#include <mavsdk/plugins/param/param.h> 

#include <chrono>
#include <functional>
#include <future>
#include <iostream>
#include <thread>

using namespace mavsdk;
using std::chrono::seconds;

void usage(const std::string& bin_name)
{
    std::cerr << "Usage : " << bin_name << " <connection_url>\n"
              << "Connection URL format should be :\n"
              << " For TCP : tcp://[server_host][:server_port]\n"
              << " For UDP : udp://[bind_host][:bind_port]\n"
              << " For Serial : serial:///path/to/serial/dev[:baudrate]\n"
              << "For example, to connect to the simulator use URL: udp://:14540\n";
}

std::shared_ptr<System> get_system(Mavsdk& mavsdk)
{
    std::cout << "Waiting to discover system...\n";
    auto prom = std::promise<std::shared_ptr<System>>{};
    auto fut = prom.get_future();

    // We wait for new systems to be discovered, once we find one that has an
    // autopilot, we decide to use it.
    mavsdk.subscribe_on_new_system([&mavsdk, &prom]() {
        auto system = mavsdk.systems().back();

        if (system->has_autopilot()) {
            std::cout << "Discovered autopilot\n";

            // Unsubscribe again as we only want to find one system.
            mavsdk.subscribe_on_new_system(nullptr);
            prom.set_value(system);
        }
    });

    // We usually receive heartbeats at 1Hz, therefore we should find a
    // system after around 3 seconds max, surely.
    if (fut.wait_for(seconds(3)) == std::future_status::timeout) {
        std::cerr << "No autopilot found.\n";
        return {};
    }

    // Get discovered system now.
    return fut.get();
}

int main(int argc, char** argv)
{
    if (argc != 2) {
        usage(argv[0]);
        return 1;
    }
    
    Mavsdk mavsdk;

    ConnectionResult connection_result = mavsdk.add_any_connection(argv[1]);

    if (connection_result != ConnectionResult::Success) {
        std::cerr << "Connection failed: " << connection_result << '\n';
        return 1;
    }

    auto system = get_system(mavsdk);
    if (!system) {
        return 1;
    }
    auto action = Action{system};
    auto telemetry = Telemetry{system};
    auto mavlink_passthrough = MavlinkPassthrough{system};
    auto mission_raw = MissionRaw{system};
    auto geofence = Geofence{system};
    auto param = Param{system};

    if (telemetry.set_rate_position(20.0) != Telemetry::Result::Success){
        std::cout << "Setting Rate failed" << std::endl;
    } else {
        std::cout << "Setting Rate succeeded" << std::endl;
    }

    while (true){
    }
}
```

julianoes, 2021-09-13:
Nice, with this example I can reproduce it! Thanks. Let's see...

julianoes, 2021-09-13:
Fix coming: https://github.com/mavlink/MAVSDK/pull/1545

PR #1545 (2021-09-14): core: fix calling callback multiple times
Fixes #1527. For review, see commits.
closed 2021-09-14 by julianoes