mavlink/MAVSDK #1531: Can MAVSdk support set ekf origin?

happyzihao, 2021-09-06:
Hello, I have a homework which needs to fly multiple UAVs, So I plan to build a global NED Coordinate. 
But when a vehicle starts to takeoff, It's current ned postion is ( 0, 0, 0 ).

I want to modify the current NED position by setting ekf_origin.

JonasVautherin, 2021-09-06:
Is there a mavlink message that is implemented in PX4 and that does that?

happyzihao, 2021-09-06:
> Is there a mavlink message that is implemented in PX4 and that does that?

Arrording to 
https://github.com/ArduPilot/MissionPlanner/issues/1901, 

I guess Mavlink should have this message.

JonasVautherin, 2021-09-06:
MAVLink messages are defined here: https://mavlink.io/en/messages/common.html

Next step is to find which one you want (if it exists) :wink:

julianoes, 2021-09-07:
Looks like it would be this one: https://mavlink.io/en/messages/common.html#SET_GPS_GLOBAL_ORIGIN

Which even seems to be implemented by PX4:
https://github.com/PX4/PX4-Autopilot/blob/5eddf32cd38f25e385a1a171319e4e08a3a4a4dc/src/modules/mavlink/mavlink_receiver.cpp#L1188-L1211

JonasVautherin, 2021-09-07:
Oh we already have a getter for that! https://github.com/mavlink/MAVSDK/blob/main/src/plugins/telemetry/telemetry_impl.cpp#L2139

happyzihao, 2021-09-07:
@julianoes @JonasVautherin , thanks for your help. Now I think mavlink message SET_GPS_GLOBAL_ORIGIN or mavsdk api TelemetryImpl::get_gps_global_origin_async may be used to set ekf_origin. 

Next, I will test this api( get_gps_global_origin_async ), and write my result.

Thanks your help.

julianoes, 2021-09-07:
The setter needs to be implemented.

See here, how a feature can be added: https://mavsdk.mavlink.io/main/en/cpp/contributing/plugins.html#add-api-to-proto

JonasVautherin, 2021-09-07:
@julianoes  Would that go into the action plugin?

julianoes, 2021-09-08:
Yes. I'd say so.

zekemyapp, 2024-04-24:
Hello! I'm trying to implement this change request for some personal test.
I wonder if this message should be in mocap instead of actions... also, should it be a sync method?

julianoes, 2024-04-26:
I think action is ok. I'd do both if both are useful.

julianoes, 2026-09-15:
This went in.

closed 2026-09-15 by julianoes

zekemyapp, 2026-09-19:
Thanks Julian!