mavlink/MAVSDK #1532: offbboard::set_position_velocity_ned

happyzihao, 2021-09-06:
Hello, 

In the funtion
Offboard::Result Offboard::set_position_velocity_ned( PositionNedYaw position_ned_yaw, VelocityNedYaw velocity_ned_yaw);

What does VelocityNedYaw mean? Average velocity in each direction ?
Actual flying, I set different VelocityNedYaw,  the fly speed is always too fast. 

Thanks.

julianoes, 2021-09-07:
It's actually a bit confusing. I believe what the speed sets is the "feed-forward" term for the control.
@Jaeyoung-Lim or @MaEtUgR maybe you can explain it better.

Jaeyoung-Lim, 2021-09-24:
@happyzihao Could you elaborate on what do you man by too fast? Is it moving faster than the desired setpoint velocity?

julianoes, 2021-09-24:
@happyzihao you can limit the speed using the param [MPC_XY_CRUISE](https://docs.px4.io/master/en/advanced_config/parameter_reference.html#MPC_XY_CRUISE).

The velocity that you add to the MAVSDK interface is a control feed-forward.

julianoes, 2021-09-24:
Note to self: we should describe that better in the API docs.

PeymanAmirii, 2022-07-28:
Thanks for your answers. In control, feed-forward usually used for disturbances or outer effects. How do you use velocity as feedforward?
For example the current point is [0, 0, 0] and the desired one is [10, 0, 0]. If we use set_position_velocity_ned(PositionNedYaw(10, 0, 0, 0),VelocityNedYaw(1, 2, 0, 0)), what happens?

julianoes, 2022-07-29:
Without getting into the details of controls, what are you trying to do?

PeymanAmirii, 2022-07-29:
I want the quadcopter to fly from its current point A to point B. I want to know what happens if vecotors AB and target velocity have different directions.

julianoes, 2022-08-02:
You want to fly from A to B, ok. And then you lost me.

PeymanAmirii, 2022-08-02:
Thanks for your answer. What I am trying to say is that we can send target velocity and target position using the command set_position_velocity_ned. Conisder that a drone is at point A and we set point B as the target point and also we consider a target velocity vector for the dorne. If we use the command set_position_velocity_ned to send the target velocity and pont but the vector from point A to B has different direction with target velocity, then what happens? 
(I think that target velocity and vector from A to B should always have the same direction, but how the command handles the case if they are not)

PeymanAmirii, 2022-08-05:
Thanks. So I think when we send both the target velocity and target position, the output is a sum of the outputs of the separate position controller and velocity controller. Therefore, they can conflict each other.

julianoes, 2022-08-05:
Right.