mavlink/MAVSDK #1534: Subscribe_position to Display Drone Position

shammaAli, 2021-09-07:
Hi,

am trying to receive drone position and display it on a QT GUI qlineedit. 
I have the code for position, but the problem when I test it using px4 I received the position once, and I need a continuous movement. The code is same as the takeoff and landing example.

`    Mavsdk mavsdk;
    ConnectionResult connection_result = mavsdk.add_any_connection("udp://:14540");

    if (connection_result != ConnectionResult::Success) {
        std::cerr << "Connection failed: " << connection_result << '\n';
        return 1;
    }
    auto system = get_system(mavsdk);
    if (!system) {
        return 1;
    }

    // Instantiate pluginis.
    auto telemetry = Telemetry{system};
    auto action = Action{system};

    // We want to listen to the altitude of the drone at 1 Hz.
    const auto set_rate_result = telemetry.set_rate_position(1);
    if (set_rate_result != Telemetry::Result::Success) {
        std::cerr << "Setting rate failed: " << set_rate_result << '\n';
        return 1;
    }

    telemetry.subscribe_position([&](Telemetry::Position position) {
        emit updateAvAlt(position.relative_altitude_m);
        emit updateAvLatLong(position.latitude_deg, position.longitude_deg);
    });`

and the emit call the display function.
could you please help?

JonasVautherin, 2021-09-07:
Your callback in `telemetry.subscribe_position([&](Telemetry::Position position) {` should be called for each position update. Is that not the case?

shammaAli, 2021-09-08:
> telemetry.subscribe_position

Does this function "telemetry.subscribe_position" call the position one time? or more than one? 
I thought setting the rate as this "const auto set_rate_result = telemetry.set_rate_position(1)" will call the position several time

closed 2024-05-27 by julianoes