mavlink/MAVSDK #1536: Linking of openssl fails on Ubuntu 18.04

julianoes, 2021-09-08:
```
cmake -DCMAKE_BUILD_TYPE=Debug -DBUILD_SHARED_LIBS=OFF -DBUILD_MAVSDK_SERVER=ON -Bbuild/debug -H.
cmake --build build/debug

[ 97%] Linking CXX executable unit_tests_mavsdk_server
../../../third_party/install/lib/libcrypto.a(dso_dlfcn.o): In function `dlfcn_load':
/home/julianoes/src/MAVSDK/build/debug/third_party/openssl/openssl/src/openssl-build/../openssl/crypto/dso/dso_dlfcn.c:116: undefined reference to `dlopen'
/home/julianoes/src/MAVSDK/build/debug/third_party/openssl/openssl/src/openssl-build/../openssl/crypto/dso/dso_dlfcn.c:119: undefined reference to `dlerror'
/home/julianoes/src/MAVSDK/build/debug/third_party/openssl/openssl/src/openssl-build/../openssl/crypto/dso/dso_dlfcn.c:138: undefined reference to `dlclose'
../../../third_party/install/lib/libcrypto.a(dso_dlfcn.o): In function `dlfcn_unload':
/home/julianoes/src/MAVSDK/build/debug/third_party/openssl/openssl/src/openssl-build/../openssl/crypto/dso/dso_dlfcn.c:161: undefined reference to `dlclose'
../../../third_party/install/lib/libcrypto.a(dso_dlfcn.o): In function `dlfcn_bind_func':
/home/julianoes/src/MAVSDK/build/debug/third_party/openssl/openssl/src/openssl-build/../openssl/crypto/dso/dso_dlfcn.c:186: undefined reference to `dlsym'
/home/julianoes/src/MAVSDK/build/debug/third_party/openssl/openssl/src/openssl-build/../openssl/crypto/dso/dso_dlfcn.c:189: undefined reference to `dlerror'
../../../third_party/install/lib/libcrypto.a(dso_dlfcn.o): In function `dlfcn_pathbyaddr':
/home/julianoes/src/MAVSDK/build/debug/third_party/openssl/openssl/src/openssl-build/../openssl/crypto/dso/dso_dlfcn.c:422: undefined reference to `dladdr'
/home/julianoes/src/MAVSDK/build/debug/third_party/openssl/openssl/src/openssl-build/../openssl/crypto/dso/dso_dlfcn.c:440: undefined reference to `dlerror'
../../../third_party/install/lib/libcrypto.a(dso_dlfcn.o): In function `dlfcn_globallookup':
/home/julianoes/src/MAVSDK/build/debug/third_party/openssl/openssl/src/openssl-build/../openssl/crypto/dso/dso_dlfcn.c:447: undefined reference to `dlopen'
/home/julianoes/src/MAVSDK/build/debug/third_party/openssl/openssl/src/openssl-build/../openssl/crypto/dso/dso_dlfcn.c:450: undefined reference to `dlsym'
/home/julianoes/src/MAVSDK/build/debug/third_party/openssl/openssl/src/openssl-build/../openssl/crypto/dso/dso_dlfcn.c:451: undefined reference to `dlclose'
collect2: error: ld returned 1 exit status
src/mavsdk_server/test/CMakeFiles/unit_tests_mavsdk_server.dir/build.make:399: recipe for target 'src/mavsdk_server/test/unit_tests_mavsdk_server' failed
make[2]: *** [src/mavsdk_server/test/unit_tests_mavsdk_server] Error 1
CMakeFiles/Makefile2:3263: recipe for target 'src/mavsdk_server/test/CMakeFiles/unit_tests_mavsdk_server.dir/all' failed
make[1]: *** [src/mavsdk_server/test/CMakeFiles/unit_tests_mavsdk_server.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
../../../third_party/install/lib/libcrypto.a(dso_dlfcn.o): In function `dlfcn_load':
/home/julianoes/src/MAVSDK/build/debug/third_party/openssl/openssl/src/openssl-build/../openssl/crypto/dso/dso_dlfcn.c:116: undefined reference to `dlopen'
/home/julianoes/src/MAVSDK/build/debug/third_party/openssl/openssl/src/openssl-build/../openssl/crypto/dso/dso_dlfcn.c:119: undefined reference to `dlerror'
/home/julianoes/src/MAVSDK/build/debug/third_party/openssl/openssl/src/openssl-build/../openssl/crypto/dso/dso_dlfcn.c:138: undefined reference to `dlclose'
../../../third_party/install/lib/libcrypto.a(dso_dlfcn.o): In function `dlfcn_unload':
/home/julianoes/src/MAVSDK/build/debug/third_party/openssl/openssl/src/openssl-build/../openssl/crypto/dso/dso_dlfcn.c:161: undefined reference to `dlclose'
../../../third_party/install/lib/libcrypto.a(dso_dlfcn.o): In function `dlfcn_bind_func':
/home/julianoes/src/MAVSDK/build/debug/third_party/openssl/openssl/src/openssl-build/../openssl/crypto/dso/dso_dlfcn.c:186: undefined reference to `dlsym'
/home/julianoes/src/MAVSDK/build/debug/third_party/openssl/openssl/src/openssl-build/../openssl/crypto/dso/dso_dlfcn.c:189: undefined reference to `dlerror'
../../../third_party/install/lib/libcrypto.a(dso_dlfcn.o): In function `dlfcn_pathbyaddr':
/home/julianoes/src/MAVSDK/build/debug/third_party/openssl/openssl/src/openssl-build/../openssl/crypto/dso/dso_dlfcn.c:422: undefined reference to `dladdr'
/home/julianoes/src/MAVSDK/build/debug/third_party/openssl/openssl/src/openssl-build/../openssl/crypto/dso/dso_dlfcn.c:440: undefined reference to `dlerror'
../../../third_party/install/lib/libcrypto.a(dso_dlfcn.o): In function `dlfcn_globallookup':
/home/julianoes/src/MAVSDK/build/debug/third_party/openssl/openssl/src/openssl-build/../openssl/crypto/dso/dso_dlfcn.c:447: undefined reference to `dlopen'
/home/julianoes/src/MAVSDK/build/debug/third_party/openssl/openssl/src/openssl-build/../openssl/crypto/dso/dso_dlfcn.c:450: undefined reference to `dlsym'
/home/julianoes/src/MAVSDK/build/debug/third_party/openssl/openssl/src/openssl-build/../openssl/crypto/dso/dso_dlfcn.c:451: undefined reference to `dlclose'
collect2: error: ld returned 1 exit status
src/mavsdk_server/src/CMakeFiles/mavsdk_server_bin.dir/build.make:186: recipe for target 'src/mavsdk_server/src/mavsdk_server' failed
make[2]: *** [src/mavsdk_server/src/mavsdk_server] Error 1
```

JonasVautherin, 2021-09-08:
Where is that? It works in the CI, doesn't it? https://github.com/mavlink/MAVSDK/blob/main/.github/workflows/main.yml#L64

julianoes, 2021-09-09:
That's in my VirtualBox and happened to @rligocki.

Abner0907, 2021-12-13:
When I run below, no errors.
> cmake -DCMAKE_BUILD_TYPE=Debug -DBUILD_SHARED_LIBS=OFF -DBUILD_MAVSDK_SERVER=ON -Bbuild/debug -H.
cmake --build build/debug

But When I build below command,
>cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=OFF -DBUILD_MAVSDK_SERVER=ON -Bbuild/default -H.
cmake --build build/default

In the fir cmake config, no errors. When build `cmake --build build/default`, the errors comes out:

>[ 96%] Built target ftp_proto_gens
[ 97%] Built target mavsdk_server
[ 97%] Linking CXX executable mavsdk_server
../../../third_party/install/lib/libcrypto.a(dso_dlfcn.o): In function `dlfcn_globallookup':
dso_dlfcn.c:(.text+0x11): undefined reference to `dlopen'
dso_dlfcn.c:(.text+0x24): undefined reference to `dlsym'
dso_dlfcn.c:(.text+0x2f): undefined reference to `dlclose'
../../../third_party/install/lib/libcrypto.a(dso_dlfcn.o): In function `dlfcn_bind_func':
dso_dlfcn.c:(.text+0x1c4): undefined reference to `dlsym'
dso_dlfcn.c:(.text+0x282): undefined reference to `dlerror'
../../../third_party/install/lib/libcrypto.a(dso_dlfcn.o): In function `dlfcn_load':
dso_dlfcn.c:(.text+0x2f1): undefined reference to `dlopen'
dso_dlfcn.c:(.text+0x369): undefined reference to `dlclose'
dso_dlfcn.c:(.text+0x3a5): undefined reference to `dlerror'
../../../third_party/install/lib/libcrypto.a(dso_dlfcn.o): In function `dlfcn_pathbyaddr':
dso_dlfcn.c:(.text+0x471): undefined reference to `dladdr'
dso_dlfcn.c:(.text+0x4d6): undefined reference to `dlerror'
../../../third_party/install/lib/libcrypto.a(dso_dlfcn.o): In function `dlfcn_unload':
dso_dlfcn.c:(.text+0x6b2): undefined reference to `dlclose'
collect2: error: ld returned 1 exit status
src/mavsdk_server/src/CMakeFiles/mavsdk_server_bin.dir/build.make:194: recipe for target 'src/mavsdk_server/src/mavsdk_server' failed
make[2]: *** [src/mavsdk_server/src/mavsdk_server] Error 1
CMakeFiles/Makefile2:2760: recipe for target 'src/mavsdk_server/src/CMakeFiles/mavsdk_server_bin.dir/all' failed
make[1]: *** [src/mavsdk_server/src/CMakeFiles/mavsdk_server_bin.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

Abner0907, 2021-12-13:
Build as below can fix this issue.
> cmake -ldl --build build/default

julianoes, 2021-12-13:
@Abner0907 thanks! How did you find that? What is the explanation?

Abner0907, 2021-12-15:
@julianoes  I google this
> In function `dlfcn_load':

and find the results, then do a try,  the error disappear.
https://stackoverflow.com/questions/26259335/error-in-libcrypto-a-when-make-c-code

julianoes, 2023-03-13:
I'm closing this as Ubuntu 18.04 is going to be EOL next month anyway.