mavlink/MAVSDK #1547: How to control velocity or fly time when using offboard::set_position_ned?

happyzihao, 2021-09-14:
Hello, when using offbaord::set_postion_ned to actual fly, the speed of vehicle is always too fast.
For safety, I want to slow down the actual fly speed.

I try to change the value of this variable: 
 const float SEND_INTERVAL_S = 0.05f;

But It's not work, How can I do ?
thanks

julianoes, 2021-09-14:
You should be able to set the speed using the param [MPC_XY_CRUISE](https://docs.px4.io/master/en/advanced_config/parameter_reference.html#MPC_XY_CRUISE).

happyzihao, 2021-09-17:
> You should be able to set the speed using the param [MPC_XY_CRUISE](https://docs.px4.io/master/en/advanced_config/parameter_reference.html#MPC_XY_CRUISE).

Thanks, It's work.

closed 2021-09-17 by happyzihao