mavlink/MAVSDK #1549: Documentation offboard is_active method is confusing

Bardo91, 2021-09-15:
According to the documentation in https://mavsdk.mavlink.io/v0.37.0/en/api_reference/classmavsdk_1_1_offboard.html#classmavsdk_1_1_offboard_1aa5e0f3c02a03f2667f82d5e162221ff5 , it is said that is_active() method returns true if offboard control is active.

> True means that the vehicle is in offboard mode and we are actively sending setpoints.

Additionally,  in section **Monitoring Offboard Mode** in https://mavsdk.mavlink.io/v0.37.0/en/guide/offboard.html . It is said

> The vehicle may change out of offboard mode outside the control of your application (for example if a GCS were to put the vehicle into Hold mode). In this case, the SDK will automatically stop sending setpoints and Offboard::is_active() will change from true to false.

From these texts, I understood that if I continuously this method and the flight_mode() from telemetry, is_active should be returning 1 if flight_mode is offboard, and 0 otherwise. 

However, I am doing a simple test and I am getting is_active() == 1 with the UAV in Altctl. This is the code.

```
    int counter = 0;
    while(run){
        std::cout <<    autopilot->mavOffboard_->is_active()  << ", " << 
                        autopilot->mavTelemetry_->flight_mode() << std::endl;

        mavsdk::Offboard::Attitude cmdRpyt{};
        cmdRpyt.roll_deg = 0;
        cmdRpyt.pitch_deg = 0;
        cmdRpyt.thrust_value = 0.2*sin(counter*0.02)+0.3;
        cmdRpyt.yaw_deg = 0;

        auto result = autopilot->mavOffboard_->set_attitude(cmdRpyt);

        auto values = autopilot->rpy();
        counter++;
        rate.wait();
    }
```

[image]

In the image, I show that is_active values 1 in Altctl, and if I pull manually the switch to toggle to offboard control, it switches the mode and the value is still 1. The engines in Altctl mode spin at a constant speed, and when I toggle to offboard I hear them swinging as expected from the code.

If I commented out the "set_attitude" command, then is when I see is_active() == 0, and obviously, the autopilot do not let me switch to offboard mode, as there are not any control command being received.
[image]

Did someone noticed it too?

If so, the documentation should say that is_active returns true when the offboard is receiving setpoint, but does not imply that the offboard mode itself is active.

julianoes, 2021-09-15:
> However, I am doing a simple test and I am getting is_active() == 1 with the UAV in Altctl.

That sounds like a bug. Thanks for reporting.

julianoes, 2023-03-13:
Hm, so reading the code, it should switch to is_active() == 0 after 3 seconds.

https://github.com/mavlink/MAVSDK/blob/e7231dc8f390b96f4682e6810ac3d2fe554b5b8b/src/mavsdk/plugins/offboard/offboard_impl.cpp#L777-L781

Could it be that you're missing that you actively need to do `offboard.start()`?

julianoes, 2025-05-06:
No more answer, closing.

closed 2025-05-06 by julianoes