mavlink/MAVSDK #1561: Why is mavlink included twice in mavsdk-windows-x64-release.zip

dzywater, 2021-09-21:
mavlink is found at
(1)mavsdk-windows-x64-release-0.43.0.zip\include\mavsdk\plugins\ftp
(2)mavsdk-windows-x64-release-0.43.0.zip\include\mavsdk\plugins\mavlink_passthrough

Both are necessary or only one or none?

julianoes, 2021-09-22:
It makes sense that the MavlinkPassthrough plugin includes mavlink because it also exports all the mavlink headers as it basically allows to send any mavlink message.

For the Ftp plugin, the public header does not include the mavlink headers from what I can see:
https://github.com/mavlink/MAVSDK/blob/main/src/plugins/ftp/include/plugins/ftp/ftp.h
I only see it included in the implementation: https://github.com/mavlink/MAVSDK/blob/14d1878024cf7531022cdc9622039988f86bfbfe/src/plugins/ftp/ftp_impl.h#L7 which I think is fine and expected as most plugins do require mavlink in the implementation.

Does that make sense? Do you see an issue with this?

dzywater, 2021-09-23:
> It makes sense that the MavlinkPassthrough plugin includes mavlink because it also exports all the mavlink headers as it basically allows to send any mavlink message.
>
> For the Ftp plugin, the public [...]

I'm afraid that I cannot undertand what you mean. 
I just think that mavlink need not to be included twice.
mavsdk-windows-x64-release-0.43.0.zip\include\mavsdk\plugins\ftp\mavlink folder can be deleted.
But I am not sure.

julianoes, 2021-09-24:
>  I just think that mavlink need not to be included twice.

It should not be.

>  mavsdk-windows-x64-release-0.43.0.zip\include\mavsdk\plugins\ftp\mavlink folder can be deleted.

Aha, now I see what you mean! Let me check why that is there. I hope we can remove it.

julianoes, 2021-09-24:
@dzywater will be fixed with https://github.com/mavlink/MAVSDK/pull/1565. Thanks for raising this.

PR #1565 (2021-09-28): ftp: remove mavlink header install
This is not required, at least not anymore.

Fixes #1561.
closed 2021-09-28 by julianoes