mavlink/MAVSDK #1566: How to connect multiple instances of mavsdk into the same mavlink port?

AlbertoUA, 2021-09-27:
Hi, I am trying to connect multiple instances of mavsdk into the same mavlink port. In order to do that, I use mavlink router to redirect the data into multiple ports. 

```
mavlink-routerd -e 127.0.0.1:29900 -e 127.0.0.1:19990 127.0.0.1:14540 
```
Then, I connect an instance of mavsdk into each new port using a **connection_url_** of udp://:19990 and udp://:29990.

```c++
    sdk_ = std::make_shared<mavsdk::Mavsdk>();

    auto prom = std::promise<std::shared_ptr<mavsdk::System>>{};
    auto fut = prom.get_future();

    mavsdk::ConnectionResult connection_result;

    connection_result = sdk_->add_any_connection(connection_url_);

    if (connection_result != mavsdk::ConnectionResult::Success) {
        RCLCPP_ERROR(this->get_logger(), "Connection failed:");
        return false;
    }

    RCLCPP_INFO(this->get_logger(), "Waiting to discover system ...");
    auto config = mavsdk::Mavsdk::Configuration(mavsdk::Mavsdk::Configuration::UsageType::CompanionComputer);
    sdk_->set_configuration(config);
    cout << connection_url_ << endl;
    sdk_->subscribe_on_new_system([this, &prom]() {
        auto system = sdk_->systems().back();
        if (system->has_autopilot()) {
            RCLCPP_INFO(this->get_logger(), "Discovered system");
            sdk_->subscribe_on_new_system(nullptr);
            prom.set_value(system);
        }
    });

    if (fut.wait_for(std::chrono::seconds(5)) == std::future_status::timeout) {
        RCLCPP_ERROR(this->get_logger(), "No Autopilot found:");
        return false;
    }
```

When I do this, only the first process to connect to a port works while the others are not able to find any system due to a timeout error.

Is there a way I can connect multiple instances of mavsdk to the same port?

JonasVautherin, 2021-09-27:
> mavlink-routerd -e 127.0.0.1:29900 -e 127.0.0.1:19990 127.0.0.1:14540 

> udp://:19990 and udp://:29990

The ports don't match here. 29900 and 29990 are not the same :innocent:

AlbertoUA, 2021-09-27:
Sorry, I mistyped the command while writing the issue. The mavlink router command I use is:
````
mavlink-routerd -e 127.0.0.1.1:29990 -e 127.0.0.0.1:19990 127.0.0.1:14540
````

JonasVautherin, 2021-09-27:
Then, for what I can see here, it looks good to me :thinking:. Did you try them separately? Is it always the first one that works, or is it always the 29990 that works?

AlbertoUA, 2021-09-28:
When I try them separately, they both work. However, it is when I try them together that the first process launched always works  and the other fails. I have found that this does not happen all the time, but randomly depending on the execution.

For example, I am trying to launch 3 instances of PX4 with ports 14540, 14541 and 14542. And then I using the mavlink router to redirect the data to various ports.

````
mavlink-routerd -e 127.0.0.1.1.1:29990 -e 127.0.0.0.1:19990 127.0.0.1.1:14540 

mavlink-routerd -e 127.0.0.0.1.1.1:29991 -e 127.0.0.0.1:19991 127.0.0.1.1:14541 

mavlink-routerd -e 127.0.0.0.1.1.1:29992 -e 127.0.0.0.1.1:19992 127.0.0.1.1:14542 
````
Next, I connect multiple instances of mavskd processes to these ports in this order

instance of process 1 connected to port 29990
instance of process 1 connected to port 29991
instance of process 1 connected to port 29992

instance of process 2 connected to port 19990
instance of process 2 connected to port 19991
instance of process 2 connected to port 19992

In this case, all instances of process 1 are running correctly. However, some instances of process 2 will randomly fail because they are not able to find a system. Failing in this if:

```c++
 if (fut.wait_for(std::chrono::seconds(5)) == std::future_status::timeout) {
        RCLCPP_ERROR(this->get_logger(), "No Autopilot found:");
        return false;
    }
````
The processes that fail are random and change between executions.

I also tried to change the second process to mavros and got the same error (some processes dies randomly sometimes because it doesn't find any system).

JonasVautherin, 2021-09-28:
One thing I can think of is that your different mavsdk instances should not have the same sysid:compid when connecting to the same drone. Not sure if that is your issue, but if not that will create issues later on, so let's try to change that first.

I would use [this](https://github.com/mavlink/MAVSDK/blob/main/src/core/mavsdk.h#L205-L206) to set manual sysid:compid (keep the boolean to false in that case)

closed 2021-09-28 by AlbertoUA