mavlink/MAVSDK #1569: Telemetry.set_rate_attitude9

fas2037, 2021-10-06:


fas2037, 2021-10-06:
I'm trying to set the rate of the attitude messages. I tried setting it to a lower hertz rate, for example 10 Hz. But the messages still seem to get published in a very high rate (100 Hz)

I subscribe to Telemetry.subscribe_attitude_quaternion()

The function should set the rate of this MAVLINK Message.
https://github.com/mavlink/MAVSDK/blob/ee84bd93b7e0ef802963503a34783772e31eb10e/src/plugins/telemetry/telemetry_impl.cpp#L215-L219

So I took a look at this:
https://github.com/mavlink/MAVSDK/blob/ee84bd93b7e0ef802963503a34783772e31eb10e/src/plugins/telemetry/telemetry_impl.cpp#L640-L720
I assume that both of this functions are able to publish the quaternion. Is there any cause for that? When checking with QGC both MAVLINK_MSG_ID_ATTITUDE_QUATERNION and MAVLINK_MSG_ID_ATTITUDE are published with 50 Hz. I guess as MAVSDK takes both messages into account and therefore the receiving rate of 100 Hz is archived.

But when using Telemetry.set_rate_attitude() I assume that this will set the rate of the subscription to a attitude message, for example the quaternion, to the desired one. This is not the case right now.

elpiel, 2021-10-31:
> My suggestion would be to change PX4 so it does not send both at 50 Hz.
Is this the default for sending telemetry messages in PX4?

In my observation, I receive telemetry responses for Position & Quarternions in 200 milliseconds from the MAVSDK server which is well below 50 Hz, I think that's 5 Hz?

I still have some evaluation to do and tweaking of the setup in order to validate how the communication is performed. I'm currently using a simulation & MAVSDK server for ease of use.

linuxguy123, 2022-07-31:
This is still a problem in MAVSDK version: v1.4.3.

julianoes, 2022-08-02:
Right. I guess we need to decouple the two attitude subscriptions and respective set_rate callers. So, for instance, instead of calling the attitude_euler callback when ATTITUDE or ATTITUDE_QUATERNION are called, only do so when ATTITUDE arrives.

julianoes, 2022-08-02:
Fix coming: 

It would be great if someone could review and test it, so I don't have to :smile:.

https://github.com/mavlink/MAVSDK-Proto/pull/293
https://github.com/mavlink/MAVSDK/pull/1878

julianoes, 2022-08-17:
:wave: anyone willing to test this? Or am I on my own here...

elpiel, 2022-08-17:
@julianoes let me try to look into this but last time I was running the sim locally I was getting bad performance. Probably because of my machine, however, I will try to test this anyway.

julianoes, 2022-08-18:
You should be able to test it even when it's running a bit slower. You can see how fast it runs in the bottom left in Gazebo, and then you need to adjust the rates that telemetry is sent by it.

julianoes, 2022-09-05:
@elpiel :wave:

elpiel, 2022-09-05:
> You should be able to test it even when it's running a bit slower. You can see how fast it runs in the bottom left in Gazebo, and then you need to adjust the rates that telemetry is sent by it.

I'm still planning to look into it, I've had some personal errands and I'm still traveling. I'll be more available in the coming 10 days.

closed 2022-09-06 by julianoes