mavlink/MAVSDK #1589: Does MAVSDK support HTTP/2?

elpiel, 2021-10-31:
Hello,
I couldn't find any mention of the supported HTTP version.

I suspect that HTTP/2 is not supported at the moment?

julianoes, 2021-10-31:
MAVSDK uses libcurl to access http. Depending on what is required, we can use different flags/features or libcurl, I believe.

Can you explain the use case, or why you would require HTTP/2?

elpiel, 2021-11-03:
When building the mavsdk client from proto files using tonic (https://github.com/hyperium/tonic/) it generates HTTP/2 enabled client.

Then I saw this article https://grpc.io/blog/grpc-on-http2/ that states:

> we’ll look at how gRPC builds on HTTP/2’s long-lived connections to create a performant, robust platform for inter-service communication. We will explore the relationship between gRPC and HTTP/2, how gRPC manages HTTP/2 connections, and how gRPC uses HTTP/2 to keep connections alive, healthy, and utilized.

I was just curious to hear if it's supported and if I can take advantage of the generated HTTP/2 client.

julianoes, 2021-11-03:
Oh, you're talking about the gRPC side! I don't actually know. I assume we would have to support a secure connection instead of the insecure one?

JonasVautherin, 2021-11-04:
I think it does use HTTP/2. What makes you think it doesn't? Does gRPC actually support HTTP/1?

elpiel, 2021-11-04:
> Oh, you're talking about the gRPC side! I don't actually know. I assume we would have to support a secure connection instead of the insecure one?

We can definitely figure out something.
Here's a suggestion:
There's is a Rust crate for end-to-end encryption called Ockam:

https://github.com/ockam-network/ockam/tree/develop/implementations/rust/ockam/ockam
There is also a C one: https://github.com/ockam-network/ockam/tree/develop/implementations/c

I've been looking into building a server-side MAVSDK in Rust.
The client is automatically built but needs some tweaking, however, using an interceptor for a Rust implemented server-side endpoint could be a PoC for end-to-end encryption.

> I think it does use HTTP/2. What makes you think it doesn't? Does gRPC actually support HTTP/1?
@JonasVautherin It does use HTTP/2, you are right. I didn't find anything mentioned in MAVSDK documentation, but I should have checked gRPC documentation too.
From my quick search it seems that it does not support HTTP/1 at all :sweat_smile:

closed 2021-11-04 by elpiel